ros-drivers/audio_common #44: rosdep install audio_common does not resolve issues for rosdep install audio_capture, etc

compiaffe, 2015-07-21:
On a Ubuntu 15.05 64-bit system running ROS Jade:

After following the tutorial here: http://wiki.ros.org/audio_common/Tutorials/Streaming%20audio 
and installing from source audio_capture had an unmet dependency on gstreamer-0.10.
##rosdep install audio_common## reported everything being installed.

However, audio_capture still had the missing dependency. Finally 
##rosdep install audio_capture## solved the issue.

Shouldn't this already happen on rosdep install audio_common?

jack-oquin, 2015-07-21:
I assume you mean 15.04, Vivid Vervet.

If you are using rosdep to resolve the dependencies, why not just install those Ubuntu packages directly?
```
sudo apt-get install ros-jade-audio-capture
```

compiaffe, 2015-07-22:
Sure, I meant 15.04.

I am modifying the source to allow for changing the topic to listen on/publish as as well as starting and stopping the audio play. (Audio_play had the same dependencies issue)

Maybe my understanding of the metapackage functionality is wrong: I thought it allows for installing or resolving of dependencies for all subpackages.

jack-oquin, 2015-07-22:
A metapackage (or any package, really) declares all its direct dependencies, allowing rosdep to install all its direct and indirect dependencies.

Exactly what dependency was installed by audio_capture, but not audio_common?

compiaffe, 2015-07-24:
I'm unsure which exact dependencies were missing... I will have to try this on a vanilla install some time next week.
Top of my head I am certain about the gstreamer-0.10 package. I'll let you know.

jack-oquin, 2015-07-24:
If you believe there is a bug, and want someone to actually change the code, please add a detailed description of how to demonstrate the error.

trainman419, 2015-08-28:
Closing because I cannot reproduce.

closed 2015-08-28 by trainman419