ros-drivers/audio_common #48: Make an Indigo Branch

DLu, 2015-08-27:
And then I will fix this:

jack-oquin, 2015-08-27:
Why not fix it in master?

mikeferguson, 2015-08-28:
master appears to be "groovy-devel" and is 52 commits behind hydro-devel. We could rename master to groovy-devel and create a new master that is hydro-devel + new stuff...

jack-oquin, 2015-08-28:
Oops, good catch, Mike. 

If groovy-devel is currently the same as master (as I suspect), I would recommend merging everything up to hydro-devel into master, then start new development with that. 

Having a down-level master branch is just confusing.

trainman419, 2015-08-28:
master synced with hydro-devel, indigo-devel branch created from current hydro branch.

closed 2015-08-28 by trainman419

trainman419, 2015-08-28:
@DLu I'm pretty sure that was fixed by #45

DLu, 2015-08-28:
Ok. It probably just isn't in the binary yet.