ros-drivers/audio_common #50: audio_play doesn't continue playing when stream restarts

BennyRe, 2015-08-28:
Here's my test case: I launch audio_play and audio_capture in separate terminals. All works fine but when I terminate audio_capture and restart it, just silence is delivered.

The same problem occurs when playing the /audio topic from a bag file.
The problem doesn't occur when I use a file sink instead the alsasink. 

The gstreamer log says nothing except
```
0:02:59.233072514 17796 0x7faecc04e120 WARN                   pulse pulsesink.c:651:gst_pulsering_stream_underflow_cb:<sink-actual-sink-pulse> Got underflow
```
when I terminate audio_capture. When I restart audio_capture the log says nothing on the info level. On the debug level it is still very verbose. I can't see a difference between the debug logs when it's working and when not. In the debug log there are messages from the source, the decoder, the converter and the sink.

The `onAudio` callback gets called as it should be and `g_signal_emit_by_name` returns successful.
I also checked that the pipeline is in PLAYING state all the time.

I checked this problem on two different machines with Ubuntu 14.04 and Indigo.

As this doesn't appear with a file sink I suspect either the decoder, converter or the sink. Does anyone have an idea how to fix this or how to further debug this?

trainman419, 2015-08-28:
I don't have the time or the expertise to investigate this. If you can find a fix for it, I'd gladly accept a pull request.

jack-oquin, 2015-08-28:
The underflow message suggests that the data are not arriving quickly enough to keep feeding your output device. That could be a bandwidth problem, or perhaps clock drift between the output device and your system time. I don't know how gstreamer handles things like that.

BennyRe, 2015-09-10:
With the fixes merged this can be closed. :+1:

PR #54 (2015-09-10): Fixed underflow.
Before the sink buffer underflows the pipeline is paused. When data is received again the pipeline is set to playing again.

This fixes #50 without changing any dependencies.
closed 2015-09-10 by BennyRe