ros-drivers/audio_common #65: Sound.repeat() does not repeat the sound

felixduvallet, 2015-11-05:
(I came across this while working on #64, personally I have no use for this functionality but I figured it would be good to report).

The SoundClient's `repeat()` method does not appear to actually repeat the sound (instead it just plays it one time).

MWE: 
```
def play_repeat():
    soundhandle = SoundClient()
    rospy.sleep(0.5)

    sound_beep = soundhandle.waveSound("say-beep.wav")
    sound_beep.play()
    rospy.sleep(1)
    sound_beep.repeat()
    rospy.sleep(5)
```

The first call plays the sound once, as expected.

The second call is expected to play the sound "repeatedly until stop() is called," but it also only plays it one time.

Tested on Ubuntu 14.04 with ros indigo.

heuristicus, 2015-12-02:
Can confirm this. I'm also running 14.04 with indigo.

zhyx490991014, 2021-07-31:
The latest version seems still has bugs, but this one works: https://github.com/yann-bourrigault/audio_common

knorth55, 2021-08-13:
In my environment with current master branch, I cannot reproduce this issue and I can repeat the wave play multiple times.

knorth55, 2021-08-13:
@zhyx490991014

can you share me your testing code?
I tested the following codes with current master branch, and it works.

```python
 import rospy
import sys

from sound_play.libsoundplay import SoundClient

def play_repeat():
    soundhandle = SoundClient()
    rospy.sleep(0.5)

    sound_beep = soundhandle.waveSound("say-beep.wav")
    sound_beep.play()
    rospy.loginfo('first beep')
    rospy.sleep(1)
    for i in range(100):
        sound_beep.repeat()
        rospy.loginfo('repeat beep: {}'.format(i))
        if rospy.is_shutdown():
            break
        rospy.sleep(1)

if __name__ == '__main__':
    rospy.init_node('repeat_test')
    play_repeat()
    sys.exit(0)
```

zhyx490991014, 2021-08-13:
Thanks for reply.
Should I run the "repeat()" in a loop?
But the "test.cpp" in Code API seems not in a loop in one test: 

```
     sleepok(2, nh);
     sound_play::Sound s1 = sc.waveSound("/usr/share/xemacs21/xemacs-packages/etc/sounds/boing.wav");
     s1.repeat();
     sleepok(1, nh);
     s1.stop();
```

> void sound_play::SoundClient::Sound::repeat	(		)	

> Play the Sound repeatedly.

> This method causes the Sound to be played repeatedly until stop() is called.

> Definition at line 98 of file sound_play.h.

I tested in Ubuntu18.04 and ROS-Melodic with current master branch and C++. 
It only play once. And it can repeat successfully when changed to this one: https://github.com/yann-bourrigault/audio_common
Here is my test code:

```
#include "ros/ros.h"
#include <sound_play/sound_play.h>

void sleepok(int t, ros::NodeHandle &nh)
 {
   if (nh.ok())
       sleep(t);
 }

int main(int argc, char **argv)
{
  ros::init(argc, argv, "talker");

  ros::NodeHandle n;

  sound_play::SoundClient sc;

  sleepok(1, n);
  // sc.startWave	(	"/home/zhyx/catkin_ws/src/sound_play/sounds/say-beep.wav", 1.0f )	;
  // sc.start(sound_play::SoundRequest::BACKINGUP);

    sound_play::Sound s1 = sc.waveSound("/home/zhyx/catkin_ws/src/sound_play/sounds/say-beep.wav");
    s1.repeat();
  sleepok(3, n);

  return 0;
}
```

knorth55, 2021-08-13:
you don't need to run `repeat` in loop, and I tried `repeat` outside of loop, and it worked as expected.
can you test my python script in your environment with `origin/master` branch?

zhyx490991014, 2021-08-13:
I tested your python script, it worked as expected.
But I tried `repeat `outside of loop, it only play once. like this:

```
#!/usr/bin/env python

import rospy
import sys

from sound_play.libsoundplay import SoundClient

def play_repeat():
    soundhandle = SoundClient()
    rospy.sleep(0.5)

    sound_beep = soundhandle.waveSound("say-beep.wav")
    sound_beep.repeat()
    rospy.loginfo('beep repeat')
    rospy.sleep(4)

if __name__ == '__main__':
    rospy.init_node('repeat_test')
    play_repeat()
    sys.exit(0)
```

By the way, I tested this in VMware Workstation. Does that matter?

knorth55, 2021-08-13:
@zhyx490991014 you forgot  `sound_beep.play()` for the first play.
this is the correct one.

```python
#!/usr/bin/env python

import rospy
import sys

from sound_play.libsoundplay import SoundClient

def play_repeat():
    soundhandle = SoundClient()
    rospy.sleep(0.5)

    sound_beep = soundhandle.waveSound("say-beep.wav")
    sound_beep.play()
    rospy.loginfo('first beep')
    rospy.sleep(3)
    sound_beep.repeat()
    rospy.loginfo('beep repeat')
    rospy.sleep(4)

if __name__ == '__main__':
    rospy.init_node('repeat_test')
    play_repeat()
    sys.exit(0)

```

zhyx490991014, 2021-08-13:
I tried your script, but it just play once when 'first beep', and another once when 'beep repeat'....

knorth55, 2021-08-13:
oh, thanks, I misunderstood about `repeat` method.
now I got your point.

knorth55, 2021-08-13:
@zhyx490991014 thank you for debugging! I merged #131 to master branch, so this issue may be solved in master branch, too.

knorth55, 2021-08-13:
this issue is solved by #131

closed 2021-08-13 by knorth55

zhyx490991014, 2021-08-13:
thank you! it worked as expected in master branch

knorth55, 2021-08-13:
Good! Thank you for helping my PR check :)