ros-drivers/audio_common #72: 'volume' error using sound_play

the8Bitts, 2016-02-03:
I can't find a way a make sound_play work. I've been searching for days already, but I can't find a fix to this error. Have any of you ever experienced this error?

`$ rosrun sound_play say.py 'Hello again'`

> Saying: Hello again Voice: voice_kal_diphone Traceback (most recent call last): File: "/opt/ros/indigo/lib/sound_play/say.py", line 79, in <module> print 'Volume: %s' % volume NameError: name 'volume' is not defined

trainman419, 2016-02-03:
I've fixed this in f3c94f5 ; can you check out the indigo-devel branch and confirm that this fixes your issue?

trainman419, 2016-02-11:
Have you had a chance to try this? If you can confirm that this patch fixes your issue, I'll do a release to binaries.

the8Bitts, 2016-02-12:
Hello! Sorry I took so long to answer.

Anyways, I couldn't get it to work by changing the code. But I managed to fix it through downloading the package with _git clone_ and using the catkin workspace instead of downloading the code with the _apt-get install ros-indigo-audio-common_.

trainman419, 2016-02-16:
This has been released into Indigo and Jade as version 0.2.11, and will be available through apt after the next sync of the build farm. The sync will be announced on the ros-users mailing list.

commit f3c94f51 (2016-02-03): Fix bug in say.py. Fixes #72
closed 2016-02-16 by trainman419