ros-drivers/audio_common #75: soundplay_node.py crashes with "Too many open files"

furushchev, 2016-06-20:
with several hours running `soundplay_node.py`, this node crashes with error "Too many open files".

```bash
(soundplay_node.py:4545): GStreamer-CRITICAL **: gst_poll_free: assertion `set != NULL' failed
Cannot connect to server socket err = No such file or directory
Cannot connect to server socket
jack server is not running or cannot be started

(soundplay_node.py:4545): GStreamer-CRITICAL **: gst_poll_write_control: assertion `set != NULL' failed

(soundplay_node.py:4545): GStreamer-CRITICAL **: gstsystemclock: write control failed in wakeup_async: 24:Too many op
en files

(soundplay_node.py:4545): GStreamer-CRITICAL **: gst_poll_free: assertion `set != NULL' failed

(soundplay_node.py:4545): GStreamer-CRITICAL **: gst_poll_write_control: assertion `set != NULL' failed

(soundplay_node.py:4545): GStreamer-CRITICAL **: gstsystemclock: write control failed in wakeup_async: 24:Too many op
en files

(soundplay_node.py:4545): GStreamer-CRITICAL **: gst_poll_free: assertion `set != NULL' failed
can't open /proc/cpuinfo
: Too many open files
```

trainman419, 2016-07-10:
There was a bug resulting in sound objects being removed from the cache but not properly closed, but that was fixed almost 4 years ago: https://github.com/ros-drivers/audio_common/commit/2657e6b535ea97f719edffd87f19505f3a29a488

Obviously the node doesn't crash if you don't play files with it; can you provide a procedure or a script that reproduces the issue?

furushchev, 2016-07-11:
@trainman419 Your fix is not enough, because in addition to python, gstreamer also has file descriptor unless `set_state(gst.STATE_NULL)` is called.

trainman419, 2016-11-10:
I have a collection of sound files in /usr/share/sounds/ubuntu/stereo that are part of the ubuntu-sounds package, so I was finally able to reproduce this by running:

```
sudo apt-get install ubuntu-sounds # just in case you don't have sound files
rosrun sound_play soundplay_node.py
```

and in a second terminal:
```
for ((i=0; i<21; i++)) ; do for f in /usr/share/sounds/ubuntu/stereo/* ; do rosrun sound_play play.py $f & sleep 0.1 ;  done ; done
```

(This would have been much faster and easier if you'd provided explicit reproduction instructions like this along with the initial pull request)

furushchev, 2017-02-28:
@trainman419 Sorry for super late reply.

> for ((i=0; i<21; i++)) ; do for f in /usr/share/sounds/ubuntu/stereo/* ; do rosrun sound_play play.py $f & sleep 0.1 ;  done ; done

I think this does not produce this issue because in this script a number of sound files is fixed.

Please test by:

```
# Terminal 1
roslaunch sound_play soundplay_node.py

# Terminal 2
for ((i=0; i<4000; i++)) ; do
  sox -n /tmp/output_$i.ogg synth 3 sin 440
  rosrun sound_play play.py /tmp/output_$i.ogg & sleep 0.1
done
```

You will find that `soundplay_node.py` on terminal exits on failure. (I needed approx. 800 files to be played.)

furushchev, 2017-03-30:
@trainman419 kindly ping to maintainer.

furushchev, 2017-04-07:
@trainman419 ping

furushchev, 2017-04-19:
@trainman419 ping

trainman419, 2017-04-19:
It seems like your test is playing a huge number of files very quickly, and is therefore filling the cache faster than it can age out old entries. Is this really representative of your usage pattern?

furushchev, 2017-04-19:
@trainman419 Yes, we create wave files using speech synthesis on each phrases and play them with `sound_play`. And I think we don't let robots speak so frequently, it should be just within normal usage...

furushchev, 2017-05-12:
@trainman419 ping

furushchev, 2017-05-24:
@trainman419 kindly ping

trainman419, 2017-05-24:
It looks like you're playing 800 files different files in 80 seconds, the cache isn't aging out the old files quickly enough.

What sort of fix would you like to see here? Do you want a parameter to decrease the cache retention time? Limit the cache size so that old entries get pushed out sooner?

furushchev, 2017-11-10:
@trainman419 Sorry for a late reply.
OK. I'll close this for now since other pull requests are blocked by this issue.
If I find any solution, I'll go back here.
Thank you very much, anyway!

closed 2017-11-10 by furushchev