ros-drivers/audio_common #76: How to use audio_capture for wave not mp3?

78226415, 2016-06-24:
I have try to use wav file for out put, then i download the source code of audio_capture, and change the code : ros::param::paramstd::string("~format", _format, "mp3"); to ros::param::paramstd::string("~format", _format, "wave"); when i run the project, it get an error: [ERROR] [1466752230.167599944]: Unsupported media type. i try to look for more information, it's return is "false" from the code : link_ok = gst_element_link_many( _source, _sink, NULL);

Could some one kindly help me? thanks.
Tags

Tags are short keywords, with no spaces within. Up to 500 tags can be used.

This field is required.

trainman419, 2016-06-24:
The source modification you're doing is changing the default for a run-time parameter; you don't need to change the source for this.

You can set parameters at run time with command line arguments, like `_format:=wave` (see http://wiki.ros.org/Remapping%20Arguments )

I'm not sure why your experiment is failing; perhaps the sample rate or the number of channels isn't set correctly. The pull request that added wave file support ( #61 ) added `capture_wave.launch`; you may want to try running that launch file:

```
roslaunch audio_capture capture_wave.launch
```

78226415, 2016-06-27:
thanks very much

closed 2016-07-10 by trainman419