ros-drivers/audio_common #79: How to change the subscriber of audio_play?

78226415, 2016-07-06:
Hi, the normal things are, it captures the audio data from audio_capture and publish the data on the topic "audio", then the audio_play subscribe the topic "audio" to get data and play it. Now, the things are changing, first it get the data from audio_capture and then do some thing to change the data and publish the changed data on a new topic, for example "audio_test", so, how to change the subscriber of audio_play to play the new topic's data?
        Why to change the data of audio_capture? first it get data from audio_capture, then change the audio data to text, maybe the text is a question, then we need to find an answer for it, when we get the answer, we should change the answer to audio and play it.
        Would any one help me? 
thanks

trainman419, 2016-07-10:
You can change which topic any node listens to by remapping the topic name; for example if you run the `audio_play` node from the command line, you can make it subscribe to the `audio_test` topic by using [remapping arguments](http://wiki.ros.org/Remapping%20Arguments):

```
rosrun audio_play audio_play audio:=audio_test
```

If you're running the mode from a launch file, you can also do this using the [`<remap>` tag](http://wiki.ros.org/roslaunch/XML/remap)

As a side note, if you have text and want to convert it to speech, the [sound_play](http://wiki.ros.org/sound_play) package may be more appropriate.

closed 2016-07-10 by trainman419

78226415, 2016-07-11:
thank you very much

78226415, 2016-07-13:
Btw, how to prepare the data of new topic for the audio_play to subscribe? I get the mp3 audio content, it is byte array, should i add the header of mp3 to the audio content?