ros-drivers/audio_common #81: How to set the versin of robotsound in sound_play?

78226415, 2016-07-26:
Hi,
   Some codes to call sound_play to play a wav file that i have writen are:
sound_play::SoundClient sc;
const char *str2 = "/home/turtlebot/Desktop/ros/Welcome.wav";
sc.startWave(str2);

then run the codes, the error is : 
[ERROR] [1469534489.657482555]: Client [/sound_play] wants topic /robotsound to have datatype/md5sum [sound_play/SoundRequest/f194e75f1c3bf7a3e1e8b16bc9ef7c2a], but our version has [sound_play/SoundRequest/d098ce4a040686259137ece23a625167]. Dropping connection.
what should i do? How to change the version of robotsound?

trainman419, 2016-07-26:
It sounds like your sound client and the soundplay node are using different versions of the SoundRequest message.

This only happens if you mix versions of sound_play from different versions of ROS.

If you're using a binary installation of ROS, please make sure that both machines are using the same ROS version.

If you have a source checkout, please make sure that it's using the correct branch.

Chloehhh, 2017-07-29:
It might because you install sound_play pkg in ros (opt/ros/...)and also save the code package(copy from github) in you workspace directory, there would be two versions of sound_play/SoundRequest msg. in your system. [one is in the ros pkg installed directory, another one is in your workspace]

If you do not `source devel/setup.bash` in your workspace before running the code, the msg file in ros directory will be requested as default, if you `#include "sound_play/SoundRequest.h" ` in your code, the msg file in your workspace package will be provided. And that is the problem...

So, you can remove one of these pkg in your system or `source workspace/devel/setup.bash` before you run the `sound_play sounplay_node.py`.

closed 2020-06-13 by knorth55