ros-drivers/audio_common #82: How to play a wav content with the bitrate of 256kbps for audio_play?

78226415, 2016-07-30:
Hi,
    we provide a wav content with the bitrate of 16kbps for the audio_play to play, it is ok. Now, if the bitrate is another one, for example 256kbps, what shoud we do? Should we change the bitrate of the wav content? Should some one help me?

thanks

jack-oquin, 2016-07-30:
Questions like this belong on http://answers.ros.org.

78226415, 2016-08-09:
I put this question to the web site http://answers.ros.org. some days ago, and no one could help me.Would you help me?  thanks

trainman419, 2016-08-09:
Actual question is here: http://answers.ros.org/question/241080/how-to-play-a-wav-content-with-the-bitrate-of-256kbps-by-audio_play/

I don't know the answer to your question; I suspect no one else has tried this.

I suspect no one has responded to your question on ROS answers because it demonstrates a lack of initiative. Your question asks "does this work," but it seems clear from the lack of answers that no one else has tried this (and they're not likely to start experimenting just to answer your question).

Perhaps if you rephrase your question as "I tried these specific things, they didn't work, and here was the error message" someone may be able to help you.

You may also want to read http://www.catb.org/~esr/faqs/smart-questions.html

Please **edit your question on ROS answers** instead of following up here.

closed 2016-08-09 by trainman419

78226415, 2016-08-11:
Hi trainman419,
    I'm sorry, it make you some uncomfortable. Now, I have added a test case with the source code to the question: http://answers.ros.org/question/241080/how-to-play-a-wav-content-with-the-bitrate-of-256kbps-by-audio_play/
Would you kindly help me? thanks