ros-drivers/audio_common #85: Official Kinetic support?

nstiurca, 2016-08-09:
Hi,

I'm wondering if you have any plans for official ROS Kinetic support (ie, binaries). I was able to compile this code from source with absolutely 0 problems under Ubuntu 16.04 + ROS Kinetic.

Thanks.

trainman419, 2016-08-10:
This is on my roadmap, but the version of audio_common that is targeted for release into Kinetic upgrades to gstreamer 1.0, and the rosdeps to depend on gstreamer are not standard across all platforms (Ubuntu vs Fedora).

There's a few hours of work required to straighten out those rosdeps before I can do a release. Now that there's interest I'll look into it.

nstiurca, 2016-08-10:
Thanks for your time, I appreciate it. Let me know if I can help with some
testing or such.

-- Nicu Știurcă (sent from my phone)

trainman419, 2016-08-14:
I've submitted the pull request to update the rosdeps needed for the Kinetic release: https://github.com/ros/rosdistro/pull/12356

Once that goes through, I'll do the release for Kinetic.

trainman419, 2016-08-28:
I've released the master branch of audio_common to kinetic: https://github.com/ros/rosdistro/pull/12525 ; if the build goes well it should be available in apt within a few weeks.

If you're interested in testing before binaries are available, feel free to check out the master branch and build it locally.

trainman419, 2016-10-06:
audio_common is now available in the public apt repo for Kinetic: http://discourse.ros.org/t/new-packages-for-kinetic-2016-10-04/544

closed 2016-10-06 by trainman419