ros-drivers/audio_common #94: interpret wav frames

arpit15, 2017-02-03:
I am trying to threshold a waveform. However the waveform I obtain with just noise has very high values. I plot a single frame. 
[image]
I can clearly see the very low levels of noise in software like audacity whose waveform is as follows
[image]
Please let me know how to get the waveform as displayed in audacity which I can use to process further.

Thank you

jack-oquin, 2017-02-04:
This question is more appropriate to http://answers.ros.org.

Just a guess: might you be plotting an unsigned value as if it were signed?

knorth55, 2020-06-13:
no response, so I will close this issue.
please feel free to reopen this.

closed 2020-06-13 by knorth55