ros-drivers/audio_common #96: Is it possible to play whole .wav file

pabiq, 2017-03-16:
I used play.py to play my own .wav file, but I discovered that it can play  around 10 seconds and stopped.
Is it possible to play longer than 10 seconds such as whole song file?

trainman419, 2017-03-16:
I'm not aware of any limitations on the length of played sounds. This sounds like a bug; if you have a sound file and a sequence of commands the reliably reproduces this behavior, I can take a closer look.

pabiq, 2017-03-26:
I just use 
$rosrun sound_play play.py 
to play my own songs, but none of them can be played longer exactly than 10 secs.

Tutorgaming, 2017-07-13:
Me too , i can only play 10 secs of the wav file

Tutorgaming, 2017-07-14:
i've found some kind of solution, it might not be the best one, but it works well on my machine.

inside the soundplay_node.py 

[soundplay_node.py](https://github.com/ros-drivers/audio_common/blob/master/sound_play/scripts/soundplay_node.py#L164
)
```python
def get_staleness(self):
.....
position = self.sound.query_position(Gst.Format.TIME)[1] # HERE change from 0 to 1
duration = self.sound.query_duration(Gst.Format.TIME)[1] # HERE change from 0 to 1 
......
except Exception, e:
position = 0
duration = 0
finally:
    self.lock.release()

if position != duration: # Seems like it checks that the seeker in end or not? 
    self.staleness = 0
else: # If the file was played until the end, Increase staleness 
    self.staleness = self.staleness + 1
    return self.staleness
```

i've found out that the value in the self.sound.query_position(Gst.Format.TIME)[0] somehow always return true , so it increases the staleness all the time until it reached the maximum value which cause the audio file purging.

Hope it helps :D

WarrenDream, 2018-08-22:
i can only play little secs of the wav file， kinetic with master branch

knorth55, 2020-06-13:
#126 solved this issue.

closed 2020-06-13 by knorth55