ros-drivers/audio_common #100: [ERROR] [1507042992.738652372]: gstreamer: Could not open audio device for recording.

piti-wity, 2017-10-03:
I am trying to use the library on my Raspberry Pi 3 that is connected to an USB mic and to headphones through jack. Anyway I have the following issue: I attach a screenshot.
[image]

Please help me, thanks in advance.

pjckoch, 2018-08-26:
Hi, I encountered the same issue on a Raspberry Pi Zero W. I tested it with the mic of the PSEye cam. Capturing audio works on my laptop, but not on the Raspi. Settings were the same. Moreover, I can capture audio on the Raspi with my own driver that uses PyAudio.

I'm wondering though: Why did you choose the title of this ticket to be "... Could not open device for recording"? I think this is misleading. You should probably call it "...Unsupported media type".

knorth55, 2020-06-13:
can you try gstreamer from cui?
also can you try with `format=wave`?
if you cannot play audio from gstreamer cui, you cannot play it in `audio_capture`, too.

knorth55, 2020-06-29:
no response, so I close this issue.
If you still have a problem, please reopen this issue

closed 2020-06-29 by knorth55

thomaspeyrucain, 2021-08-17:
I had the same issue when launching : `roslaunch audio_capture capture.launch`
To fix it, make sure you have all this packages up to date :
`apt-get install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libgstreamer-plugins-bad1.0-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-doc gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-alsa gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-qt5 gstreamer1.0-pulseaudio`
Then check the configuration of your microphone with :
`arecord -l` To know what to put for hw:<card_number>,<device_number>
And with pulseaudio installed :
`pactl list short sinks` To know the number of channels, the sample_format and the sample_rate

Then if you launch with the correct config it should work :
`roslaunch audio_capture capture.launch channels:=2 sample_rate:=44100 device:=hw:1,0`

Check the audio topic to see if it is correctly working:
`rostopic echo /audio/audio`