ros-drivers/audio_common #104: audio_play | no output when capturing using wave format

warp1337, 2018-01-24:
If I start capturing like the following:

<pre>
roslaunch audio_capture capture.launch format:=wave
</pre>

And playing sound like:

<pre>
roslaunch play.launch
</pre>

All I get is:
<pre>
process[audio/audio_play-1]: started with pid [31732]
[ WARN] [1516804179.851736153] /audio/audio_play::RosGstPlay::cb_need_data: need-data signal emitted! Pausing the pipeline
[ WARN] [1516804180.081790523] /audio/audio_play::RosGstPlay::cb_need_data: need-data signal emitted! Pausing the pipeline
[ WARN] [1516804180.180440179] /audio/audio_play::RosGstPlay::cb_need_data: need-data signal emitted! Pausing the pipeline
[ WARN] [1516804180.181167783] /audio/audio_play::RosGstPlay::cb_need_data: need-data signal emitted! Pausing the pipeline
</pre>

The rostopic is published with approx. 100 hz and 32 kb/bandwidth
Also, when setting format to mp3 everything works fine.

satan-ruler, 2018-09-05:
audio_capture.cpp
#if 0
          caps = gst_caps_new_simple("audio/x-raw",
                                     "channels", G_TYPE_INT, _channels,
                                     "width",    G_TYPE_INT, _depth,
                                     "depth",    G_TYPE_INT, _depth,
                                     "rate",     G_TYPE_INT, _sample_rate,
                                     "signed",   G_TYPE_BOOLEAN, TRUE,
                                     NULL);
#else
          caps = gst_caps_from_string("audio/x-raw,format=S16LE,rate=16000,channels=1,layout=interleaved");
#endif

satan-ruler, 2018-09-05:
audio_play.cpp

//g_object_set(G_OBJECT(_source), "do-timestamp", TRUE, NULL);
//gst_bin_add( GST_BIN(_pipeline), _source);
...
else if (_format == "wave")
{
	GstCaps *caps;
	caps = gst_caps_from_string("audio/x-raw,format=S16LE,rate=16000,channels=1,layout=interleaved");
	g_object_set( G_OBJECT(_source), "caps", caps, NULL);
	gst_caps_unref(caps);
	g_object_set (G_OBJECT (_source), "format", GST_FORMAT_TIME, NULL);
	_sink = gst_element_factory_make( "autoaudiosink", "sink" );
	gst_bin_add_many( GST_BIN(_pipeline), _source, _sink, NULL);
	gst_element_link_many( _source, _sink, NULL);
}

knorth55, 2020-06-13:
I make PR for this issue #127

knorth55, 2020-06-29:
#127 is merged, so please try it.

closed 2020-06-29 by knorth55