ros-drivers/audio_common #106: apt-get install sound_play package failure in Debian Jessie

AllenSimon, 2018-04-04:
I'd like to install sound_play package in my debian jessie arm64 board,however I couldn't install it by using "$apt-get install ros-kinetic-sound-play" because of unmet dependencies "libgstreamer-plugins-good1.0-0" ,the detail is listed as below:

```
root@linaro-alip:~/catkin_ws/src/# apt-get install ros-kinetic-sound-play
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-kinetic-sound-play : Depends: libgstreamer-plugins-good1.0-0 but it is not installable
E: Unable to correct problems, you have held broken packages.

```
The linux distribution I used was Debian Jessie arm64, ros distribution was kinetic.I installed the sound_play package because I'd like to test some speech recognition and synthesis demos on my company's boards. I had installed and tested some demos on ubuntu 14.04 armhf (ros indigo) on another board,these demos mostly needed the support of the sound_play package,there were no problems on it. 

I needed to run these demos on our new 64bits Debian Jessie boards.I checked the wiki and make sure ROS Kinetic supported Debian Jessie.However,the error log showed that "libgstreamer-plugins-good1.0-0" could't be installed in Debian.

In fact,I checked it in both official websites of debian and ubuntu,the package "libgstreamer-plugins-good1.0-0" indeed didn't exist in Debian's package repository : https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=libgstreamer-plugins-good1.0-0,while it exist in ubuntu's package repository:https://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=libgstreamer-plugins-good1.0-0&searchon=names

so,I have some problems:
1.How to solve the dependency problem in Debian Jessie?
2.Are there any other similar audio packages available that supported Debian?
3.Can I make a little change of the source codes to make the sound_play package support Debian?

Thanks!

gavanderhoorn, 2018-04-04:
According to the OP, building from source on Jessie arm64 does work.

Perhaps the `rosdep` rules for [gstreamer1.0-plugins-good](https://github.com/ros-drivers/audio_common/blob/0b1878f3c182cd20b5bd3c8f202d63f5b1fc44ca/sound_play/package.xml#L31-L34) need arm64 specific mappings ([here](https://github.com/ros/rosdistro/blob/05cb828d194ac6eec0f02fc286948747f56d964c/rosdep/base.yaml#L1040-L1042)?

AllenSimon, 2018-04-04:
I tried the command "$ apt-get install ros-kinetic-sound-play" both on Debian Jessie arm64 and amd64, they showed the same problem.
I solved this problem on Jessie amd64, but I couldn't verify it on my arm64 board now beacuse the rosdep failure on it...I guessed it might result from the proxy problem which was ignored by me before.

gavanderhoorn, 2018-04-04:
So is the installation failure a result of a proxy, or because the libraries are really not available / installable on Jessie ARM64?

AllenSimon, 2018-04-04:
No,the installation failure isn't a result of a proxy, I chose to use "$ apt-get " to install sound_play on my board because I couldn't use rosdep.
I tried to intall sound_play on both amd64(it had proxy problem) and arm64(it didn't have the problem),the error logs showed the same :"libgstreamer-plugins-good1.0-0" is not installable. I checked in official website , the libraries are really not available / installable on all distributions of Debian.

gavanderhoorn, 2018-04-04:
Which `deb` packages did you install using `apt-get` on jessie amd64?

AllenSimon, 2018-04-04:
??What did you mean?
As the error log showed,the sound_play package needed the dependency "libgstreamer-plugins-good1.0-0",so I "$apt-get install libgstreamer-plugins-good1.0-0" but failed beacuse it's not  available .

AllenSimon, 2018-04-04:
I just would like to explain that if there isn't proxy problem,I can run sound_play by building it from source.
But if I insist on using the command "$ apt-get install ros-kinetic-sound-play" ,I'll still fail.

gavanderhoorn, 2018-04-04:
@AllenSimon: according to [packages.debian.org/jessie/gstreamer1.0-plugins-good](https://packages.debian.org/jessie/gstreamer1.0-plugins-good) the `gstreamer1.0-plugins-good` package does exist. And that is also what the [gstreamer1.0-plugins-good](https://github.com/ros/rosdistro/blob/05cb828d194ac6eec0f02fc286948747f56d964c/rosdep/base.yaml#L1096-L1101) key maps to at the moment.

Can you verify that `apt-get install gstreamer1.0-plugins-good` succeeds? Note, there is no `lib` prefix.

gavanderhoorn, 2018-04-04:
Looking at `debian/control` of `ros-kinetic-sound-play`, the `libgstreamer-plugins-*` dependencies seem strange:

```
Depends: ${shlibs:Depends}, ${misc:Depends}, [...], gstreamer1.0-plugins-base,
  gstreamer1.0-plugins-good, gstreamer1.0-plugins-ugly, gstreamer1.0-tools,
  libgstreamer-plugins-base1.0-0, libgstreamer-plugins-good1.0-0,
  libgstreamer1.0-0, [...]
```

As Debian (Jessie) doesn't have those, they make `apt` error out upon resolution of the dependencies.

I'm not a maintainer here (or author), so I'll leave it at this.

AllenSimon, 2018-04-09:
@gavanderhoorn 
hi,I can install "gstreamer1.0-plugins-good" and "gstreamer-plugins-good1.0-0" by "apt-get",and in fact,I have installed them before.But it still made no sense.
So ,do you mean that if I'd like to use the sound_play package, I only have two choices?
1.Build the package from source instead of using “$ apt-get”.
2.Wait for the author to fix the bug,and re-release the driver and insist on using the command
“apt-get install ros-kinetic-sound-play”.
The last update time of the package was 2016......I think the possibility is quite low.

But I cannot use the command  "rosdep" to build sound_play package now beacuse of the proxy problem in my company.Can I build it without using "rosdep" in another way?

AllenSimon, 2018-05-18:
I can install sound_play package by the command "$ apt-get install ros-kinetic-sound-play" in my Debian Jessie arm64 board now, it seems that the bug has been fixed. Thanks!

closed 2018-05-18 by AllenSimon

gavanderhoorn, 2018-05-18:
I don't see any commits recently merged that influence the build infra side of things, so perhaps something has changed on Jessie's side.

trainman419, 2018-05-22:
I did a new release recently, so that probably picked up new rosdep package definitions or something.