ros-drivers/audio_common #108: Soundplay_node.py not working. Error

spyderroque, 2018-05-16:
Hello, I am new to ROS and to Sound_play.

When I try to execute anything related to sound_play like 
`$rosrun sound_play say.py 'hello' 
I receive an error message in the terminal running soundplay_node.py
Message is:
`command: 1
volume: 1.0
arg: "hello"
arg2: "voice_kal_diphone"
[ERROR] [1526462894.456404]: Exception in callback: global name 'gst' is not defined
[INFO] [1526462894.457356]: Traceback (most recent call last):
  File "/home/dave/catkin_ws/src/audio_common/sound_play/scripts/soundplay_node.py", line 286, in callback
    sound = self.select_sound(data)
  File "/home/dave/catkin_ws/src/audio_common/sound_play/scripts/soundplay_node.py", line 248, in select_sound
    self.voicesounds[data.arg] = soundtype(wavfilename, self.device, data.volume)
  File "/home/dave/catkin_ws/src/audio_common/sound_play/scripts/soundplay_node.py", line 80, in __init__
    self.sound = gst.element_factory_make("playbin2",None)
NameError: global name 'gst' is not defined`

Is this an error in the file or am I doing something wrong?

trainman419, 2018-05-16:
This looks like it might be a bug that was introduced into sound_play recently.

Which ROS release are you using, and what does `rosversion sound_play` show as the version for the sound_play package?

spyderroque, 2018-05-17:
Hi,

I am using ROS Kinetic on a 16.04 LTS Ubuntu Unity.
`rosversion` tells me I am using version 0.3.2.
I am wondering: Would the previous version 0.3.1. work with kinetic?

tony1213, 2018-05-22:
Hi,
    
    I am new to audio,python,spending 2 days on this ROS package.
    You can change this file soundplay_node.py at line 80 and 85.
    
    Change line 80 
    self.sound = gst.element_factory_make("playbin2",None)
    To
    self.sound = Gst.ElementFactory.make("playbin",None)

    Change line 85 
    self.sink = gst.element_factory_make("alsasink", "sink")
    To
    self.sink = Gst.ElementFactory.make("alsasink", "sink")
    
    Then Success.
    
    Tips: My OS is Ubuntu 16.04 LTS,using ROS Kinetic.

trainman419, 2018-05-22:
It looks like this was broken by https://github.com/ros-drivers/audio_common/commit/482b09ec88fbbe9ecec97b726e9b737c41077197 .

closed 2018-05-22 by trainman419

trainman419, 2018-05-22:
Fixed in 0.3.3: https://github.com/ros/rosdistro/pull/17940 ; this should show up in the public packages the next time the ROS maintainers do a sync for kinetic, lunar and melodic.

JanhviK, 2018-06-13:
@tony1213 The file soundplay_node.py is in opt directory & hence it's a "READ-ONLY" file . So how can i change from gst to Gst?

JanhviK, 2018-06-13:
@spyderroque  Dude was ur problem solved ?? Because I'm getting the same error as urs.

trainman419, 2018-06-13:
@JanhviK  I've confirmed that this fix has been synced to all of the public deb repositories, so you should be able to fix this by upgrading the audio-common package through apt-get. If you need specific commands to run, please list your Ubuntu and ROS versions.

JanhviK, 2018-06-13:
@trainman419 Yeah finally it worked, i changed the configurations of the file and the changes were made appropriately. Even gstreamer was missing, so it was sorted by downloading its package.

commit 3752fea3 (2018-05-22): Fix gstreamer errors. Fixes #108