ros-drivers/audio_common #109: soundplay_node.py crashes after a certain period

janbogdanov, 2018-05-22:
Hello

I have a ros node which successfully executes the following lines of (pseudo)code:

```
sc.playWaveFromPkg("sound_play", "/home/jan/synthts_et-master/synthts_et/out_tnu.wav");
sleep(seconds + 1);
sc.stopAll();
```

I get the desired sound for approximately 300 iterations or so (which corresponds to hours of audio). However, after that my soundplay_node.py crashes with the following error message:

```
[ERROR] [1524690159.260685]: Exception in callback: Environment configuration is invalid: cannot locate rospack ([Errno 24] Too many open files)
[INFO] [1524690159.261880]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 273, in callback
    sound = self.select_sound(data)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 208, in select_sound
    absfilename = os.path.join(roslib.packages.get_pkg_dir(data.arg2), data.arg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslib/packages.py", line 188, in get_pkg_dir
    raise InvalidROSPkgException("Environment configuration is invalid: cannot locate rospack (%s)"%e)
InvalidROSPkgException: Environment configuration is invalid: cannot locate rospack ([Errno 24] Too many open files)
```

The only nodes running are the soundplay_node.py and the node publishing the .wav file. I have contemplated on this issue and my own thoughts have halted to a standstill. Maybe you have some insight into this. Thank you!

gavanderhoorn, 2018-05-22:
Follow-up of [soundplay_node.py crashes after a certain period](https://answers.ros.org/question/291768) on ROS Answers.

trainman419, 2018-05-22:
This sounds like https://github.com/ros-drivers/audio_common/issues/75 and https://github.com/ros-drivers/audio_common/issues/7 are back.

These have traditionally been very difficult issues for me to reproduce locally. Please provide a complete ROS package and step-by-step instructions to reproduce this.

janbogdanov, 2018-05-23:
It seems #75 and I indeed have similar issues. 

The system is attempting to implement speech synthesis and basically comprises of the following elements:

1) A publisher node, which publishes a simple message (`ss << "Hello, world " << count;`) at a rate of  (`ros::Rate loop_rate(0.1);`)

2) A subscriber node, which subscribes to the messages (`ros::Subscriber sub = n.subscribe("chatter", 1000, chatterCallback);`), performs speech synthesis on it in the subscriber callback (it takes approximately 100ms), after which a .wav file is created. That .wav file is then played using sound_play in the way described above.

3) The soundplay_node.py node, which is "hot-fixed" for another problem with a solution shown in #96 

The system is using ROS-Kinect and sound_play 0.3.1.

If I am failing to provide the information you were expecting, please let me know. Thank you in advance!

itohdak, 2019-10-08:
Hello, is this issue still ongoing?
We have also been facing the same issue .

After running `soundplay_node.py` for a while, we get the following errors, which is almost the same as #75.
```
(python:5789): GStreamer-CRITICAL **: gst_poll_free: assertion 'set != NULL' failed

(python:5789): GStreamer-CRITICAL **: gst_poll_write_control: assertion 'set != NULL' failed

(python:5789): GStreamer-CRITICAL **: gstsystemclock: write control failed in wakeup_async: 24:Too many open files
```
or the last one can be
```
(python:5789): GStreamer-CRITICAL **: gstsystemclock: write control failed in wakeup_async: 2:No such file or directory
```

Following the above errors comes a different type of errors.
```
Lexicon: compile file "/usr/share/festival/dicts/wsj.wp39.poslexR" not found or unreadble 
[ERROR] [1554100428.699768]: Sound synthesis failed. Is festival installed? Is a festival voice installed? Try running "rosdep satisfy sound_play|sh". Refer to http://wiki.ros.org/sound_play/Troubleshooting
[ERROR] [1554100428.701407]: Exception in actionlib callback: 'NoneType' object has no attribute 'command'
[INFO] [1554100428.702542]: Traceback (most recent call last):
  File "/home/itohdak/audio_debug_ws/src/ros-drivers/audio_common/sound_play/scripts/soundplay_node.py", line 373, in execute_cb
    sound.command(data.command)
AttributeError: 'NoneType' object has no attribute 'command'

[WARN] [1554100428.705807]: Your executeCallback did not set the goal to a terminal status.  This is a bug in your ActionServer implementation. Fix your code!  For now, the ActionServer will set this goal to aborted
```

I think the second errors occur due to the first errors; GStreamer errors.
I tried to fix this problem in #126 .

I would appreciate it if you could leave some comments.

knorth55, 2020-06-13:
i merge #126 , so this issue should be solved.
if you still have the issue, please feel free to reopen this

closed 2020-06-13 by knorth55