ros-drivers/audio_common #112: print() is a function in Python 3

cclauss, 2018-07-12:
flake8 testing of https://github.com/ros-drivers/audio_common on Python 3.6.3

$ __flake8 . --count --select=E901,E999,F821,F822,F823 --show-source --statistics__
```
./sound_play/scripts/play.py:44:58: E999 SyntaxError: invalid syntax
        print 'Usage: %s sound_to_play.(ogg|wav) [volume]'%sys.argv[0]
                                                         ^
./sound_play/scripts/playbuiltin.py:43:45: E999 SyntaxError: invalid syntax
        print 'Usage: %s <sound_id> [volume]'%sys.argv[0]
                                            ^
./sound_play/scripts/playpackage.py:44:66: E999 SyntaxError: invalid syntax
        print 'Usage: %s package sound_to_play.(ogg|wav) [volume]'%sys.argv[0]
                                                                 ^
./sound_play/scripts/say.py:44:44: E999 SyntaxError: invalid syntax
        print 'Usage: %s \'String to say.\''%sys.argv[0]
                                           ^
./sound_play/scripts/shutup.py:49:50: E999 SyntaxError: invalid syntax
    print "Sending stopAll commande every 100 ms."
                                                 ^
./sound_play/scripts/soundplay_node.py:62:13: E999 SyntaxError: invalid syntax
    print str
            ^
./sound_play/scripts/test.py:58:107: E999 SyntaxError: invalid syntax
    print "This script will run continuously until you hit CTRL+C, testing various sound_node sound types."
                                                                                                          ^
./sound_play/scripts/test_actionlib_client.py:15:27: E999 SyntaxError: invalid syntax
    print "Need Unplugging"
                          ^
8     E999 SyntaxError: invalid syntax
8
```

trainman419, 2018-07-15:
audio_common is in maintenance mode. Until ROS requires python3, I won't have personal time to fix this. Feel free to submit a pull request if you would like this fixed sooner.

knorth55, 2020-06-13:
I fixed this issue for noetic release.

closed 2020-06-13 by knorth55