ros-drivers/audio_common #120: audio_play stops playing after a couple of seconds

nicolapiccinelli, 2019-05-28:
I'm trying to build an audio bridge between ROS and Unity3D using ROS#. I'm currently able to stream the audio from Unity3D to ROS and play it back correctly using the audio_play and a raw2encoded custom node. But I'm experiencing strange behavior from ROS to Unity3D (like huge cracking and low volume).

So I've tried to simply run the **audio_capture** and the **audio_play**, as shown in the tutorial and I've found that the audio_play stops playing after a couple of seconds (every time). Any ideas?

I run the nodes like this
```
roslaunch audio_capture capture.launch
roslaunch audio_play play.launch
```

Moreover, it seems that if i run the *capture_wave.launch* the topic isn't playable by the **audio_play** node. The output of the audio_play node when the playback stops is the following:
```
0:00:01.566186985 11707       0xa6a4f0 WARN           audiobasesink gstaudiobasesink.c:1807:gst_audio_base_sink_get_alignment:<sink-actual-sink-pulse> Unexpected discontinuity in audio timestamps of -0:00:00.190000000, resyncing
0:00:01.566764887 11707 0x7fd860003890 WARN                   pulse pulsesink.c:702:gst_pulsering_stream_underflow_cb:<sink-actual-sink-pulse> Got underflow
0:00:01.636297236 11707       0xa6a4f0 WARN           audiobasesink gstaudiobasesink.c:1807:gst_audio_base_sink_get_alignment:<sink-actual-sink-pulse> Unexpected discontinuity in audio timestamps of +0:00:00.000250000, resyncing
0:00:04.193932984 11707       0xa6a4f0 WARN           audiobasesink gstaudiobasesink.c:1807:gst_audio_base_sink_get_alignment:<sink-actual-sink-pulse> Unexpected discontinuity in audio timestamps of +0:00:00.003125000, resyncing
```

I'm able to reproduce correctly the sound using the following command:
```
gst-launch-1.0 alsasrc device=hw:0,0 ! audioconvert ! audioresample ! alsasink
```

quattrinili, 2019-05-31:
I just made a pull request that should fix your problem (at least I had the same problem and that fixed it), please see #122 .

In practice, I added a parameter for changing the value of `do-timestamp`: it should be false for livestream audio.

nicolapiccinelli, 2019-05-31:
I will try!

nicolapiccinelli, 2019-06-05:
@quattrinili i've solved the issue removing all the references to gstreamer 0.1

closed 2020-06-13 by knorth55