ros-drivers/audio_common #124: soundplay_node.py can not find any music files.

botamochi6277, 2019-07-31:
When I executed `soundplay_node.py` and `test.py` , `soundplay_node.py` did not play musics and showed following error messages. 

I found "/opt/ros/kinetic/share/sound_play/sounds/say-beep.wav" in my Ubuntu. 
I can play it with `aplay` command. 

```bash
aplay /opt/ros/kinetic/share/sound_play/sounds/say-beep.wav 
xcb_connection_has_error() returned true
xcb_connection_has_error() returned true
Playing WAVE '/opt/ros/kinetic/share/sound_play/sounds/say-beep.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
```

## Executed commands when I got errors.

[Terminal1]
```bash
rosrun sound_play test.py
```

[Terminal2]
```bash
rosrun sound_play soundplay_node.py _device:="hw:2,0"
[INFO][/sound_play][1564544453.756883]: sound_play node is ready to play sound
[ERROR][/sound_play][1564544466.486080]: Error setting up to play "/opt/ros/kinetic/share/sound_play/sounds/say-beep.wav". Does this file exist on the machine on which sound_play is running?
[ERROR][/sound_play][1564544466.489885]: Exception in callback: 'NoneType' object has no attribute 'command'
[INFO][/sound_play][1564544466.494558]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 287, in callback
    sound.command(data.command)
AttributeError: 'NoneType' object has no attribute 'command'

[ERROR][/sound_play][1564544468.600545]: Error setting up to play "/opt/ros/kinetic/share/sound_play/sounds/say-beep.wav". Does this file exist on the machine on which sound_play is running?
[ERROR][/sound_play][1564544468.605016]: Exception in callback: 'NoneType' object has no attribute 'command'
[INFO][/sound_play][1564544468.608707]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 287, in callback
    sound.command(data.command)
AttributeError: 'NoneType' object has no attribute 'command'

[ERROR][/sound_play][1564544470.601662]: Exception in callback: 'NoneType' object has no attribute 'set_property'
[INFO][/sound_play][1564544470.606020]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 286, in callback
    sound = self.select_sound(data)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 264, in select_sound
    self.builtinsounds[data.sound] = soundtype(params[0], self.device, volume)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 86, in __init__
    self.sink.set_property("device", device)
AttributeError: 'NoneType' object has no attribute 'set_property'

[ERROR][/sound_play][1564544472.604559]: Exception in callback: 'NoneType' object has no attribute 'set_property'
[INFO][/sound_play][1564544472.608272]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 286, in callback
    sound = self.select_sound(data)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 264, in select_sound
    self.builtinsounds[data.sound] = soundtype(params[0], self.device, volume)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 86, in __init__
    self.sink.set_property("device", device)
AttributeError: 'NoneType' object has no attribute 'set_property'

[ERROR][/sound_play][1564544474.606188]: Exception in callback: 'NoneType' object has no attribute 'set_property'
[INFO][/sound_play][1564544474.609911]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 286, in callback
    sound = self.select_sound(data)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 264, in select_sound
    self.builtinsounds[data.sound] = soundtype(params[0], self.device, volume)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 86, in __init__
    self.sink.set_property("device", device)
AttributeError: 'NoneType' object has no attribute 'set_property'

[ERROR][/sound_play][1564544476.608999]: Exception in callback: 'NoneType' object has no attribute 'set_property'
[INFO][/sound_play][1564544476.613249]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 286, in callback
[... 78 lines cut ...]
    self.sink.set_property("device", device)
AttributeError: 'NoneType' object has no attribute 'set_property'

sound: -3
command: 0
volume: 1.0
arg: "Testing the new A P I"
arg2: ''
[ERROR][/sound_play][1564544496.431277]: Exception in callback: 'NoneType' object has no attribute 'set_property'
[INFO][/sound_play][1564544496.435209]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 286, in callback
    sound = self.select_sound(data)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 248, in select_sound
    self.voicesounds[data.arg] = soundtype(wavfilename, self.device, data.volume)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 86, in __init__
    self.sink.set_property("device", device)
AttributeError: 'NoneType' object has no attribute 'set_property'

```

## My Environment
ubuntu version: Ubuntu 16.04.5 LTS
rosversion: kinetic
sound_play version: 0.3.3
hardware: RaspberryPi 3 B+

lotharchris, 2020-02-02:
I am having the same exact issue on my pi 3 B+, with a USB speaker.
Were you able to resolve your issue?

botamochi6277, 2020-02-03:
I have not solve this problem yet. 
I had another problem that `ros-soundplay` cannot change a speaker in raspberry Pi. 
To solve this, I developed [original package](https://github.com/botamochi6277/sound_pipes) using aplay.

lotharchris, 2020-02-04:
Thanks for the link to the package, but I cannot clone it.I get a Warning, then a fatal.Not sure if I can resolve this, or if you need to do something.Any direction is appreciated.

$ git clone git@github.com:botamochi6277/sound_pipes.git
Cloning into 'sound_pipes'...
The authenticity of host 'github.com (192.30.253.112)' can't be established.
RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
Are you sure you want to continue connecting (yes/no)? y
Please type 'yes' or 'no': yes
Warning: Permanently added 'github.com,192.30.253.112' (RSA) to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

lotharchris, 2020-02-04:
Nevermind.
I got it to work using https
git clone https://github.com/botamochi6277/sound_pipes.git

knorth55, 2020-06-12:
>[ERROR][/sound_play][1564544493.035874]: Exception in callback: 'NoneType' object has no attribute 'set_property'
[INFO][/sound_play][1564544493.038833]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 286, in callback
    sound = self.select_sound(data)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 264, in select_sound
    self.builtinsounds[data.sound] = soundtype(params[0], self.device, volume)
  File "/opt/ros/kinetic/lib/sound_play/soundplay_node.py", line 86, in __init__
    self.sink.set_property("device", device)
AttributeError: 'NoneType' object has no attribute 'set_property'

this problems is fixed by adding missing dependency #142 
if you still have problem, feel free to reopen this issue.

closed 2020-06-12 by knorth55