ros-drivers/audio_common #125: sound_play: can't play more than 32 files

ryanewel, 2019-08-14:
It looks like sound_play can't handle getting audio from more than 32 different files while running.

OS: Ubuntu 16.04
ROS Distro: Kinetic 
Installation source: apt

Steps to reproduce:
1) Create copies of the sound files
e.g.
```
cd /opt/ros/kinetic/share/sound_play/sounds
for i in `seq 0 32`; do
  sudo cp "NEEDS_PLUGGING.ogg" "NEEDS_PLUGGING_${i}.ogg"
done
```
2) Launch the sound_play node
```
roslaunch sound_play soundplay_node.launch
```
3) Send files to sound play
```
# Based on test.py and test_actionlib.py
import rospy, os, sys
import roslib; roslib.load_manifest('sound_play')
from sound_play.msg import SoundRequest

from sound_play.libsoundplay import SoundClient

if __name__ == '__main__':
    rospy.init_node('soundplay_test', anonymous = True)
    client = SoundClient(blocking=True)

    base_path = os.path.join(roslib.packages.get_pkg_dir('sound_play'),'sounds')
    for i in range(32):
        path = base_path + "/NEEDS_PLUGGING_{}.ogg".format(i)
        print path
        client.playWave(path)
    path = base_path + "/NEEDS_PLUGGING_BADLY.ogg"
    print path
    client.playWave(path)
    print "All audio files played"
```

4) Output is:
```
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_0.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_1.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_2.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_3.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_4.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_5.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_6.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_7.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_8.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_9.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_10.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_11.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_12.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_13.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_14.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_15.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_16.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_17.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_18.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_19.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_20.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_21.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_22.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_23.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_24.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_25.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_26.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_27.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_28.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_29.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_30.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_31.ogg
/opt/ros/kinetic/share/sound_play/sounds/NEEDS_PLUGGING_32.ogg

```
And then it just hangs. Audio output comes through except for the last one. (Can test by changing range to 31 and noticing that NEEDS_PLUGGING_BADLY isn't played).

The number is reset after restarting the node.

knorth55, 2020-06-13:
@ryanewel does this issue still occurs?
we merged #126 , so I hope the problem is solved.

knorth55, 2020-06-29:
no response, so i  close this issue.
If you still have problem, please reopen this issue.

closed 2020-06-29 by knorth55