ros-drivers/audio_common #130: Add or change maintainers on this repository

clalancette, 2020-03-20:
@trainman419 I see that you are currently the maintainer of this repository, but may not have time for it anymore.  Would you mind if we added another one or two maintainers to this repository?  In particular, we'd like to add @k-okada as a co-maintainer here (and maybe one other person).  See the discussion at https://discourse.ros.org/t/ros-drivers-audio-common-bug-fixes/13302 .

Thanks in advance!

trainman419, 2020-03-21:
Yeah; I haven't really had time to manage the bug reports or contributions on the repository for a while. I've granted @k-okada and @knorth55 write access to this repository and https://github.com/ros-gbp/audio_common-release so that they can do maintenance and releases.

closed 2020-03-21 by trainman419

knorth55, 2020-03-23:
Thank you so much!