ros-drivers/audio_common #132: Release audio_common into ROS Noetic?

sloretz, 2020-04-23:
Would you mind releasing `audio_common` to ROS Noetic? It looks like all of its dependencies have been released, and recursively 16 repos need it.

I'm not sure how much work it will need. The [Noetic Migration Guide](http://wiki.ros.org/noetic/Migration) or the [guide about transitioning ROS packages to Python 3](http://wiki.ros.org/UsingPython3/) may be useful.

knorth55, 2020-04-23:
@sloretz thank you, I'm working on it. #133 
I'm not sure about python3 migration, so I need to search more.
i.e.
if we have `catkin_install_python`, we don't need `install` anymore?
I found following line in https://github.com/ros/catkin/blob/noetic-devel/doc/howto/format2/installing_python.rst, but I'm not perfectly sure what it means.
```
Note: As in setuptools, the packages list is not recursive, and sub-packages must be included explicitly (e.g. your_package.tools.my_util which would contain the python modules defined in the folder src/your_package/tools/my_util/, along with an __init__.py file).
```

sloretz, 2020-04-24:
> if we have catkin_install_python, we don't need install anymore?

Correct. `catkin_install_python()` installs the script with a rewritten shebang.

> I found following line in https://github.com/ros/catkin/blob/noetic-devel/doc/howto/format2/installing_python.rst, but I'm not perfectly sure what it means.

I think that's referring to the [`packages` keyword argument to `setup()`](https://packaging.python.org/guides/distributing-packages-using-setuptools/#packages). I don't think that note matters here since there is one Python package: `sound_play`.

sloretz, 2020-04-29:
https://github.com/ros/rosdistro/pull/24624

closed 2020-04-29 by sloretz