ros-drivers/audio_common #136: Add new audio message type with more information than audio_common_msgs/AudioData

708yamaguchi, 2020-04-30:
Hi ROS communities,

When I use `audio_common_msgs/AudioData` message, I am sometimes confused because there is no field describing audio information, such as channels sampling rate.
Also, currently, non-compressed audio data and compressed audio data is converted into the same message `audio_common_msgs/AudioData`.

In this issue, I would like to discuss adding new audio message type. Here, I propose 2 new message types.
I am wodering backward compatibility and understandability of these messages. I would like to get your feedback.

#### 1. AudioData.msg:
(If backward compatibility is important, other name (e.g. WaveData.msg) is better)
```
Header header  # Header timestamp should be acquisition time of audio
               # Header frame_id should be position of microphone

uint8 channels # Number of channels
uint32 rate    # Sampling rate
uint8 bitdepth # Number of bits per audio data

uint8[] data   # Audio data buffer
               # WAVE format is assumed
```

The first message is for non-compressed audio data. I think WAVE format is suitable because existing programs supports WAVE format. For example, https://github.com/ros-drivers/audio_common/blob/fece72eab8559b2dd3f206179896d52ded17e5cf/audio_capture/src/audio_capture.cpp#L109
For creating this message, I referred to this page.
http://soundfile.sapp.org/doc/WaveFormat/
In this format, I do not use redundant data field, such as byte rate.

#### 2. CompressedAudioData.msg:
```
Header header  # Header timestamp should be acquisition time of audio
               # Header frame_id should be position of microphone

uint32 bitrate # Amount of audio data per second [bits/s]

string format  # Specifies the format of the audio data
               # For example, mp3
uint8[] data   # Compressed audio buffer
```

The second message is for compressed audio data. I assume mp3 format because existing programs supports mp3 format. For example, https://github.com/ros-drivers/audio_common/blob/fece72eab8559b2dd3f206179896d52ded17e5cf/audio_capture/src/audio_capture.cpp#L98
I add bitrate field so that users can calculate how many seconds each audio data contains.
(In the first message, `rate` * `bitdepth` * `channels` equals to `bitrate`)

I create these two message types with reference to `sensor_msgs/Image` and `sensor_msgs/CompressedImage`
Image: http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Image.html
CompressedImage: http://docs.ros.org/melodic/api/sensor_msgs/html/msg/CompressedImage.html

Thanks,

knorth55, 2020-05-01:
Hi, thank you for the proposal.
I think it is good to add new audio msg type with meta data and also compressed audio msg type.
However, for backward compatibility, maybe it is good to add in new name space, IMO.
Also, I want more discussion for this issue, so please give us your idea here.

rokusottervanger, 2020-06-16:
I completely agree with this proposal. I'm not sure what the backward compatibility issues would be, but I'll leave that up to the maintainer(s). I'll close #91 as this covers the usability issue.

708yamaguchi, 2020-06-17:
Thank you very much for your comments.

I created a pull request https://github.com/ros-drivers/audio_common/pull/148

knorth55, 2020-12-17:
#148 is merged

closed 2020-12-17 by knorth55