ros-drivers/audio_common #137: sound_play cannot find msg files due to .egg creation

GuilhermePSC, 2020-05-11:
### My Environment

Ubuntu xenial 16.04.6 LTS
rosversion: kinetic
sound_play version: 0.3.5

### Issue

**sound_play** tries to locate the `msg` folder in `/opt/ros/robot/lib/python2.7/dist-packages/sound_play-0.3.5-py2.7.egg`, which is created in **0.3.5**, but fails (`ImportError: No module named msg`) to do so because it only contains:

[image]

while it should locate msg here: `/opt/ros/robot/lib/python2.7/dist-packages/sound_play/msg`

The cause is the [change introduced in 0.3.5](https://github.com/ros-drivers/audio_common/commit/d26805e48210fa43ff1a2371e4f82d111147e423) from `distutils.core`, which doesn't generate the .egg, to `setuptools`, which does, and the issue appears to happen only for xenial (and older?)

knorth55, 2020-05-11:
Hi, thank you for reporting the issue.
I tried with `osrf/ros:kinetic-desktop-xenial` docker image, but I cannot reproduce your error.
can you tell me how to reproduce your error?
```
root@897387d714f2:~# python
Python 2.7.12 (default, Oct  8 2019, 14:14:10) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from sound_play.msg import SoundRequest
>>> msg = SoundRequest()
>>> msg
sound: 0
command: 0
volume: 0.0
arg: ''
arg2: ''
>>> exit()
root@897387d714f2:~# ls -R /opt/ros/kinetic/lib/python2.7/dist-packages/sound_play
sound_play/                sound_play-0.3.5.egg-info/ 
root@897387d714f2:~# ls -R /opt/ros/kinetic/lib/python2.7/dist-packages/sound_play*
/opt/ros/kinetic/lib/python2.7/dist-packages/sound_play:
__init__.py  __init__.pyc  libsoundplay.py  libsoundplay.pyc  msg

/opt/ros/kinetic/lib/python2.7/dist-packages/sound_play/msg:
_SoundRequest.py        _SoundRequestAction.pyc          _SoundRequestActionGoal.py    _SoundRequestActionResult.pyc  _SoundRequestGoal.py    _SoundRequestResult.pyc
_SoundRequest.pyc       _SoundRequestActionFeedback.py   _SoundRequestActionGoal.pyc   _SoundRequestFeedback.py       _SoundRequestGoal.pyc   __init__.py
_SoundRequestAction.py  _SoundRequestActionFeedback.pyc  _SoundRequestActionResult.py  _SoundRequestFeedback.pyc      _SoundRequestResult.py  __init__.pyc

/opt/ros/kinetic/lib/python2.7/dist-packages/sound_play-0.3.5.egg-info:
PKG-INFO  dependency_links.txt  top_level.txt
```

knorth55, 2020-06-13:
no response, and it seems the issue is strongly related to the environment, so I will close this issue.
if you find out this issue is `audio_common` problem, please feel free to reopen this.

closed 2020-06-13 by knorth55