ros-drivers/audio_common #138: systemd not working properly with audio

roboticsai, 2020-05-17:
I have two ros launch files one for master/controller and another for robot/slave. Problem i'm facing is when i run the scripts or launch files directly everything working fine i.e the audio is comming properly in earphone which i'm using to test the audio play. But when i run this scrips containing the launch files with systemd audio is not comming. here is the master.service:

[Unit]
Description=master node

[Service]
Type=simple
ExecStart=/home/odroid/scripts/startup_launch.sh
Restart=on-failure

[Install]
WantedBy=multi-user.target
slave0.service

[Unit]
Description=slave0 node

[Service]
Type=simple
ExecStart=/home/odroid/scripts/startup_launch.sh
Restart=on-failure

[Install]
WantedBy=multi-user.target
~/scripts/startup_launch.sh of master:

#!/bin/bash
source /opt/ros/melodic/setup.bash
source /home/odroid/catkin_ws/devel/setup.bash
export ROS_IP=192.168.1.111
export ROS_MASTER_URI=http://192.168.1.111:11311
roslaunch catkin_pkg catkin.launch
~/scripts/startup_launch.sh of slave:

[Unit]
Description=slave0 node

[Service]
Type=simple
ExecStart=/home/odroid/scripts/startup_launch.sh
Restart=on-failure

[Install]
WantedBy=multi-user.target
catkin.launch file of master:

<launch>
    <node pkg="rosserial_arduino" type="serial_node.py" name="publisher" respawn="true">
        <param name="port" value="/dev/ttyACM0"/>
    </node>

    <include file="$(find audio_play)/launch/play.launch">
        <arg name="do_timestamp" value="false" />
        <arg name="ns" value="cap1" />
    </include>

</launch>
catkin.launch file of slave0:

<launch>
    <include file="$(find audio_capture)/launch/capture.launch">
        <arg name="device" value="hw:1,0"/>
        <arg name="ns" value="cap1" />
    </include>
    <node pkg="rosserial_arduino" type="serial_node.py" name="subscriber"/>
    <node pkg="cv_camera" type="cv_camera_node" name="cam1"/>
</launch>
Here when i launch the scrips or roslaunch file directly from terminal frist in master than in slave the audio is comming properly, except if the slave launch file is terminated and started audio don't comes unless i again restart the master launch file.

I'm using ros melodic, odorid c2 both with ubuntu 18.04 mate version in them.

roswtf
shows not error or warnings. the rostopic list is:

/cam1/camera_info
/cam1/image_raw
/cam1/image_raw/compressed
/cam1/image_raw/compressed/parameter_descriptions
/cam1/image_raw/compressed/parameter_updates

/cam1/image_raw/compressedDepth
/cam1/image_raw/compressedDepth/parameter_descriptions
/cam1/image_raw/compressedDepth/parameter_updates
/cam1/image_raw/theora
/cam1/image_raw/theora/parameter_descriptions
/cam1/image_raw/theora/parameter_updates
/cap1/audio
/diagnostics
/rosout
/rosout_agg
/twist
Here /cap1/audio is the audio data, which is comming ok/properly when i echo is with rostopic

echo /cap1/audio

cmd.

knorth55, 2020-06-13:
it seems the issue is strongly related to the environment, so I will close this issue.
if you find out this issue is `audio_common` problem, please feel free to reopen this.

closed 2020-06-13 by knorth55

knorth55, 2023-02-07:
I found out this issue is related to `pulseaudio` `systemd`.
In order to run `audio_common` from `systemd`, we need to set `XDG_RUNTIME_DIR` properly.
if you want to run `audio_common` without `login`, you need to run `pulseaudio` in system wide mode, not user mode.
`pulseaudio`'s default is used mode.

knorth55, 2023-02-07:
please check #219 for more detailed info