ros-drivers/audio_common #139: roslaunch not launching nodes properly

roboticsai, 2020-05-17:
I'm getting problem while launching the audio play nodes from launch file. When i run the audio play node from terminal it works for first time, i.e if i stop this node and than again run it from terminal the audio is not comming in earphone.

when i launch the audio play from the launch file the sound dont' comes at all. When i check if the audio data is comming or not by using rostopic echo "/audio/audio" audio data are being published. Here the the audio play launch file which i run in master odroid c2.

`<launch> <node pkg="rosserial_arduino" type="serial_node.py" name="publisher" port="/dev/ttyACM0"/> <include file="$(find audio_play)/launch/play.launch" &gt;="" <param="" name="do_timestamp" value="false"/> <arg name="ns" value="cap1"/> </include> </launch>`

audio capture launch file:

```
<launch>
    <include file="$(find audio_capture)/launch/capture.launch">
        <param name="device" value="hw:1,0"/>
    </include>
</launch>
```
I'm running two odrid c2 for this purpose to make a robot for my small project. Here one is master running play( controller ) and another is slave ( robot ). i'm using tp link wifi dongle in robot which connected to router connected to controller. i also don't know how to debug this issue since their is no presenece of erros or warnings. + the nodes and running properly.

knorth55, 2020-06-13:
is the audio topic really published?
can you check it with `rostopic hz`?
can the topic be subscribed from the other computer?
please check those points.

we want more report for debugging this issue.

roboticsai, 2020-06-20:
Yes, when i check it with rostopic hz and echo, the data is comming. And this problem is only occuring in raspberry pi. I.e their is no audio output.
In odroid c2, it is working properly. In my personal computer also it is working.

knorth55, 2020-12-17:
sorry for late response. I'm busy these days.
Do you have any update for this issue?

knorth55, 2021-08-13:
if you still have the issue, please reopen this issue.

closed 2021-08-13 by knorth55