ros-drivers/audio_common #142: Incorrect dependencies

5tan, 2020-06-08:
After installing `ros-melodic-audio-common` in docker/ubuntu18.04, sound output didn't work (no error in console, similar issue: https://answers.ros.org/question/263077/sound_play-has-no-sound-and-no-errors/).

I got sound working after installing additional packages: 
```
sudo apt install libgstreamer1.0-0 gstreamer1.0-pulseaudio gstreamer1.0-libav gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad
```

knorth55, 2020-06-08:
How can you reproduce this error?
Which node are you trying?

5tan, 2020-06-08:
> How can you reproduce this error?

I was following tutorial: http://wiki.ros.org/sound_play/Tutorials/ConfiguringAndUsingSpeakers
First: `rosrun sound_play soundplay_node.py`
Then: `rosrun sound_play say.py "hello world"`

> How can you reproduce this error?

Install `ros-melodic-audio-common` in some minimal VM, container or RPi. In my case there was no sound until I have installed those additional packages.

knorth55, 2020-06-09:
can you give us commands to run your docker image?
I'm checking in my environment as below now.

```
docker  run --device /dev/snd     -e PULSE_SERVER=unix:${XDG_RUNTIME_DIR}/pulse/native     -v ${XDG_RUNTIME_DIR}/pulse/native:${XDG_RUNTIME_DIR}/pulse/native     -v ~/.config/pulse/cookie:/root/.config/pulse/cookie     --group-add $(getent group audio | cut -d: -f3)     -it osrf/ros:melodic-desktop-bionic /bin/bash

# inside docker
apt update; apt install tmux ros-melodic-audio-common
# install pulseaudio
apt install pulseaudio
# i install these packages below, and it works
apt install gstreamer1.0-alsa
```

5tan, 2020-06-09:
> can you give us commands to run your docker image?
unfortunately no :( 

So you need to install manually `pulseaudio` and `gstreamer1.0-alsa`. Those packages are not listed as a `audio_common` package dependency, because other OS might require different "sound framework"?

knorth55, 2020-06-09:
>unfortunately no :(

so, unfortunately, we cannot figure out your error is specific to your environment or not.
we can use either `pulseaudio` or `alsa`, it is not required.
however, `gstreamer1.0-alsa` is missing package dependency.

knorth55, 2020-06-12:
I open a PR #144

knorth55, 2020-06-12:
if you still have problem, please feel free to reopen this issue.

closed 2020-06-12 by knorth55

iPwnXX, 2021-04-29:
I still can not solve the problem even after installing these dependencies.  no sound and no error, but `aplay` works