ros-drivers/audio_common #143: Linking CXX executable ERROR

noelia-dica, 2020-06-11:
When I try to compile this package with catkin_make -j1, I get the following error:

odroid@odroid:~/catkin_ws$ catkin_make -j1
Base path: /home/odroid/catkin_ws
Source space: /home/odroid/catkin_ws/src
Build space: /home/odroid/catkin_ws/build
Devel space: /home/odroid/catkin_ws/devel
Install space: /home/odroid/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/odroid/catkin_ws/build"
####
####
#### Running command: "make -j1" in "/home/odroid/catkin_ws/build"
####
[  0%] Built target _audio_common_msgs_generate_messages_check_deps_AudioData
[  1%] Built target audio_common_msgs_generate_messages_nodejs
[  2%] Built target audio_common_msgs_generate_messages_cpp
[  3%] Built target audio_common_msgs_generate_messages_lisp
[  6%] Built target audio_common_msgs_generate_messages_py
[  9%] Built target audio_common_msgs_generate_messages_eus
[  9%] Built target audio_common_msgs_generate_messages
[  9%] Built target std_msgs_generate_messages_lisp
[  9%] Built target std_msgs_generate_messages_py
[  9%] Built target std_msgs_generate_messages_nodejs
[  9%] Built target roscpp_generate_messages_cpp
[  9%] Built target roscpp_generate_messages_eus
[  9%] Built target roscpp_generate_messages_nodejs
[  9%] Built target roscpp_generate_messages_py
[  9%] Built target rosgraph_msgs_generate_messages_cpp
[  9%] Built target std_msgs_generate_messages_eus
[  9%] Built target roscpp_generate_messages_lisp
[  9%] Built target rosgraph_msgs_generate_messages_eus
[  9%] Built target rosgraph_msgs_generate_messages_py
[  9%] Built target rosgraph_msgs_generate_messages_nodejs
[  9%] Built target rosgraph_msgs_generate_messages_lisp
[  9%] Built target std_msgs_generate_messages_cpp
[ 10%] Linking CXX executable /home/odroid/catkin_ws/devel/lib/audio_capture/audio_capture
CMakeFiles/audio_capture.dir/src/audio_capture.cpp.o: In function `main':
audio_capture.cpp:(.text+0x64): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
CMakeFiles/audio_capture.dir/src/audio_capture.cpp.o: In function `audio_transport::RosGstCapture::RosGstCapture()':
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCaptureC2Ev[_ZN15audio_transport13RosGstCaptureC5Ev]+0x30): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCaptureC2Ev[_ZN15audio_transport13RosGstCaptureC5Ev]+0x6bc): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCaptureC2Ev[_ZN15audio_transport13RosGstCaptureC5Ev]+0x77e): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCaptureC2Ev[_ZN15audio_transport13RosGstCaptureC5Ev]+0x83a): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCaptureC2Ev[_ZN15audio_transport13RosGstCaptureC5Ev]+0x8fc): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCaptureC2Ev[_ZN15audio_transport13RosGstCaptureC5Ev]+0xb36): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCaptureC2Ev[_ZN15audio_transport13RosGstCaptureC5Ev]+0xbf8): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCaptureC2Ev[_ZN15audio_transport13RosGstCaptureC5Ev]+0xc76): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCaptureC2Ev[_ZN15audio_transport13RosGstCaptureC5Ev]+0xd38): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
CMakeFiles/audio_capture.dir/src/audio_capture.cpp.o: In function `audio_transport::RosGstCapture::onMessage(_GstBus*, _GstMessage*, void*)':
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCapture9onMessageEP7_GstBusP11_GstMessagePv[_ZN15audio_transport13RosGstCapture9onMessageEP7_GstBusP11_GstMessagePv]+0x80): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
audio_capture.cpp:(.text._ZN15audio_transport13RosGstCapture9onMessageEP7_GstBusP11_GstMessagePv[_ZN15audio_transport13RosGstCapture9onMessageEP7_GstBusP11_GstMessagePv]+0x148): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
CMakeFiles/audio_capture.dir/src/audio_capture.cpp.o: In function `bool ros::param::param<std::string>(std::string const&, std::string&, std::string const&)':
audio_capture.cpp:(.text._ZN3ros5param5paramISsEEbRKSsRT_RKS4_[_ZN3ros5param5paramISsEEbRKSsRT_RKS4_]+0xe): undefined reference to `ros::param::has(std::string const&)'
audio_capture.cpp:(.text._ZN3ros5param5paramISsEEbRKSsRT_RKS4_[_ZN3ros5param5paramISsEEbRKSsRT_RKS4_]+0x1c): undefined reference to `ros::param::get(std::string const&, std::string&)'
CMakeFiles/audio_capture.dir/src/audio_capture.cpp.o: In function `bool ros::param::param<int>(std::string const&, int&, int const&)':
audio_capture.cpp:(.text._ZN3ros5param5paramIiEEbRKSsRT_RKS4_[_ZN3ros5param5paramIiEEbRKSsRT_RKS4_]+0xe): undefined reference to `ros::param::has(std::string const&)'
audio_capture.cpp:(.text._ZN3ros5param5paramIiEEbRKSsRT_RKS4_[_ZN3ros5param5paramIiEEbRKSsRT_RKS4_]+0x1c): undefined reference to `ros::param::get(std::string const&, int&)'
CMakeFiles/audio_capture.dir/src/audio_capture.cpp.o: In function `void ros::Publisher::publish<audio_common_msgs::AudioData_<std::allocator<void> > >(audio_common_msgs::AudioData_<std::allocator<void> > const&) const':
audio_capture.cpp:(.text._ZNK3ros9Publisher7publishIN17audio_common_msgs10AudioData_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN17audio_common_msgs10AudioData_ISaIvEEEEEvRKT_]+0x86): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
audio_capture.cpp:(.text._ZNK3ros9Publisher7publishIN17audio_common_msgs10AudioData_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN17audio_common_msgs10AudioData_ISaIvEEEEEvRKT_]+0x192): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
audio_capture.cpp:(.text._ZNK3ros9Publisher7publishIN17audio_common_msgs10AudioData_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN17audio_common_msgs10AudioData_ISaIvEEEEEvRKT_]+0x286): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
audio_capture.cpp:(.text._ZNK3ros9Publisher7publishIN17audio_common_msgs10AudioData_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN17audio_common_msgs10AudioData_ISaIvEEEEEvRKT_]+0x39c): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
audio_capture.cpp:(.text._ZNK3ros9Publisher7publishIN17audio_common_msgs10AudioData_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN17audio_common_msgs10AudioData_ISaIvEEEEEvRKT_]+0x4a8): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
[... cut ...]
audio_common/audio_capture/CMakeFiles/audio_capture.dir/build.make:119: recipe for target '/home/odroid/catkin_ws/devel/lib/audio_capture/audio_capture' failed
make[2]: *** [/home/odroid/catkin_ws/devel/lib/audio_capture/audio_capture] Error 1
CMakeFiles/Makefile2:974: recipe for target 'audio_common/audio_capture/CMakeFiles/audio_capture.dir/all' failed
make[1]: *** [audio_common/audio_capture/CMakeFiles/audio_capture.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed

If I compile my catkin_ws with my other packages I have no errors, what can I do?

knorth55, 2020-06-11:
it seem `ros` is not correctly imported.
did you source `/opt/melodic/ros/setup.bash` ?

noelia-dica, 2020-06-12:
@knorth55 Yes I did.
It worked perfectly until I installed libuvc_camera, as it gave an error compiling I eliminated it and I used this command: $sudo apt-get purgue --auto-remove libuvc. Then I did catkin_make and nothing worked. 
The thing is that when audio_common is not into my catkin_ws, catkin_make works and compiles properly with my other packages but the compiling fails when audio_common package is in the catkin_ws.
Do you know what could I do, I tried to follow what they said here: https://answers.ros.org/question/220579/undefined-reference-to-rosinitint-char-stdstring-const-unsigned-int/?answer=220602#post-id-220602 but it didn't work.

knorth55, 2020-06-12:
hmm, you purged something important or some config got changed.
can you try this and reinstall dependencies, remove old build files and rebuild?
```
rosdep install --ignore-src --from-paths <catkin_ws_path>/src -r -y
```

noelia-dica, 2020-06-12:
I tried this:
$rosdep install --ignore-src --from-paths /home/odroid/catkin_ws/src -r -y
It said: All required rosdep installed successfully.
$sudo apt install python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential
Then I removed old build files and the catkin_make gave me exactly the same error.

Can you think of anything else I can do, please?

knorth55, 2020-06-12:
can you tell me which distribution you are using?
can you also check `ros-<distro>-rosconsole` is installed in your environment?

noelia-dica, 2020-06-12:
I am using melodic and yes, rosconsole is installed.

noelia-dica, 2020-06-12:
It is Melodic, and yes, rosconsole is installed.

knorth55, 2020-06-13:
No idea to solve your problem, sorry for that.

try these things below
1. remove the workspace, create it again, and build it
2. create a new workspace which only has `audio_common` and build it
3. stop using current environment (reinstall whole system or use docker).

also, the issues seems not `audio_common` problem but your environment problem, so I will close this issue now.
if you find out if it is related to `audio_common`, feel free to reopen this.

P.S.
my advice is that your issue seems the environment setting, so it is better to ask your friend personally for more quick debugging.

closed 2020-06-13 by knorth55

noelia-dica, 2020-06-18:
Finally I removed ROS, I intalled it again following the tutorial. I did $sudo apt-get install ros-melodic-audio-common, I put my other packages in catkin_ws and now it works perfectly.
Thankyou for your help!!!