ros-drivers/audio_common #153: Feature Request: Support device by id

d-walsh, 2020-08-13:
In other ROS device drivers it is possible to specify the device by id - which is very useful on systems that have multiple microphones connected. When I tried to pass the device id string to `audio_capture` it gave the error:
```
[/audio/audio_capture]: ros.audio_capture: gstreamer: Could not open audio device for recording.
```
The device that I want to use is: `/dev/snd/by-id/usb-xxxx-2314322-00`

knorth55, 2020-08-14:
can you check this decice feature?
https://github.com/ros-drivers/audio_common/pull/87

it uses not usb device id but alsa device id.

knorth55, 2021-09-01:
no response, so I will close this issue.
Feel free to reopen this issue.

closed 2021-09-01 by knorth55