ros-drivers/audio_common #157: audio_capture: gstreamer internal data flow error

Kyuhae, 2020-09-29:
Hi,

When I run audio_capture with default parameters, it uses my laptop microphone, no issues.
When I run audio_capture with a different device parameter (i.e. hw:1,0 to use some high quality mics via a DAW), the audio_capture node crashes immediately with a "**gstreamer internal data flow error"**.
The ros logs for the audio_capture node are empty / not present.
I can use the microphones fine via the command line:
gst-launch-1.0 alsasrc ! audioconvert ! audioresample ! alsasink
and recording from the mics works fine with ffmpeg.

**I seem to have fixed this** by setting the device I want as the default ALSA device (creating an alsa config file to select default device), and passing an empty string as the device parameter value in my audio_capture launchfile.

Is this normal? why would it crash when specifying the device, but not when passing the empty string?

If you need any more info to check or reproduce i'll be happy to help out.

knorth55, 2020-09-29:
which version are you using? the current origin/master? or debian package?
I think we can run audio_capture with custom device parameters, probably.
can you tell me how to reproduce your problem?

Kyuhae, 2020-10-02:
> which version are you using? the current origin/master? or debian package?
I'm using the current origin/master branch.

> I think we can run audio_capture with custom device parameters, probably.
I was able to run audio_capture with this device specified in the launchfile before, (i think it was with the debian package, 6 months ago). If I specify my laptop internal microphone (set device parameter as "hw:0,0"), it works fine. If i specify the device corresponding to my external microphones (ser device parameter as "hw:1,0"), it systematically crashes with the error i mentioned: "gstreamer internal data flow error".

> can you tell me how to reproduce your problem?
I took the default audio_capture launch file, made my own launchfile based on it, just changing parameter values (device, sample rate, channels, format). Works fine with empty device parameter, works fine with specifying "hw:0,0", and crashes with "hw:1:0". And as i said, it works with my external mics after setting those as the default ALSA device, and then passing the empty device parameter to audio_capture. Obviously, I changed the parameters such as sample rate and channels in accordance to which mic it was supposed to be using.

Since it works with the laptop mic (hw:0,0), and not with the external mic (hw:1,0), I'm not too sure. Might depend on the microphone... 
When I say "external mic", it's  two steinberg mics (XLR plugs) connected to an audio interface, which then connects to the pc via USB.
So I suppose the first step is to check with ANY other mic you have. I also have a simpler mic that i could try, if you like (that just plugs in via a standard jack to the pc).

knorth55, 2020-10-06:
@Kyuhae Hi, I'm sorry to reply to you late. I'm currently busy, but I will figure out the error.
If you have time, you can debug by yourself and make a PR.
We are welcoming your contribution to `audio_common`.

Kyuhae, 2020-10-09:
It's ok, for now this workaround of changing the default device and passing empty parameter works for me. I know nothing about gstreamer API, and don't really have the time to debug now either. But if I get the time and find a solution, for sure i'll make a PR.

knorth55, 2020-10-14:
are you getting `internal data flow error` or internal data stream error`?
can you past whole error and launch output?
also can you try with `plughw1,0`?
also can you check the output of `gst-device-monitor-1.0`?
can you try these commands below?
```
# autoaudiosrc
gst-launch-1.0 -v autoaudiosrc num-buffers=1000 device="hw:1,0" ! audio/x-raw,channels=1,depth=16,rate=16000 ! avimux ! filesink location=test.avi
# alsasrc
gst-launch-1.0 -v alsasrc num-buffers=1000 device="hw:1,0" ! audio/x-raw,channels=1,depth=16,rate=16000 ! avimux ! filesink location=test.avi

```

knorth55, 2020-10-14:
also you can get your mic's original sample rate from the command below:

```bash
pactl list short sinks
```

maybe most of mic's sample rate is 44100 Hz, so can you try with `sample_rate:=44100` or command below?

```bash
gst-launch-1.0 -v autoaudiosrc num-buffers=1000 device="hw:1,0" ! audio/x-raw,channels=1,depth=16,rate=44100 ! avimux ! filesink location=test.avi
```

knorth55, 2020-10-21:
I found out that we need to correctly set all parameters if you set `device:=hw:1,0`.
The parameters are `channels`, `sample_rate`, `depth` and so on.
If you use `device:=plughw:1,0`, you can avoid this error, probably.

knorth55, 2020-12-02:
No response, so I close this issue.
Feel free to reopen this issue if you still have the same problem.

closed 2020-12-02 by knorth55