ros-drivers/audio_common #158: Audio_capture not from a device, but straing from wav file (source is a .wav file not microphone)

antondim, 2020-10-06:
Hello, as title suggests I'm trying to use a pre-recorded .wav file as audio source and publish it's data as a signed int16[] or byte[] audio stream, to /audio (or /audio/audio) topic. 

Documentation seems to ignore this case, even though it's stated in description of "audio_capture" package.

antondim, 2020-10-08:
could this be a node that's raised inside a python script?

knorth55, 2020-10-08:
what do you mean? I don't get what you wan to say.
`audio_capture` node is written in c++.
https://github.com/ros-drivers/audio_common/blob/master/audio_capture/src/audio_capture.cpp

antondim, 2020-10-08:
It would be easier for me to manipulate audio (devices, stream, data) through python, so I was thinking to call an extra python node via the audio capture launchfile . This node would publish the stream data from python's pyaudio module to a separate topic etc etc . But ok, will look into it in my free time