ros-drivers/audio_common #164: python version of audio_play does not work

ChanganVR, 2020-12-15:
Thank you for this repo. I'm using python to play the audio data published by audio_capture, but somehow the captured data only produces white noise mp3 files. Am I missing any post-processing steps? The script is attached below. Any suggestion is appreciated!

```
#!/usr/bin/env python
import rospy
from audio_common_msgs.msg import AudioData
import wave

def callback(msg, queue):
    queue.append(msg.data)

rospy.init_node('listener', anonymous=True)
channel = []
rospy.Subscriber("audio/audio", AudioData, callback, channel)
rospy.sleep(2)

wf = wave.open('wave.mp3', 'wb')
wf.setnchannels(1)
wf.setsampwidth(2)
wf.setframerate(16000)
wf.writeframes(b''.join(channel))
wf.close()
```

knorth55, 2020-12-16:
i have no idea about `wave` package because we uses `gstreamer` in this package.
I recommend to use `audio_play` node to subscribe audio topics and save them in `mp3` if you quickly need a solution.
https://github.com/ros-drivers/audio_common/blob/master/audio_play/src/audio_play.cpp#L58-L123

You can save audio topics to `mp3` by changing `dst` argument to file path.
https://github.com/ros-drivers/audio_common/blob/master/audio_play/launch/play.launch

also, you can use `gstreamer` in python, so you can simply rewrite our node to python, too.

ChanganVR, 2020-12-16:
Hi @knorth55 thank you for you answer. I took a look at the audio_play.cpp and it looked like that file basically writes the input frames directly to a file in mp3 format. And below is my modifed script in case other people need it. 

```
#!/usr/bin/env python
import rospy
from audio_common_msgs.msg import AudioData

def callback(msg, queue):
    queue.append(msg.data)

rospy.init_node('listener', anonymous=True)
frames = []
rospy.Subscriber("audio/audio", AudioData, callback, frames)
rospy.sleep(2)

file = open("audio.mp3", "wb")
file.write(b''.join(frames))
file.close()
```

closed 2020-12-16 by ChanganVR

ChanganVR, 2020-12-17:
I just gained more understanding of why wave wasn't able to write the correct signal while directly writing raw data to files works and I wanted to confirm it with you @knorth55. I think the reason is that the received msg.data contains some meta information of each data block (gst.caps) and treating them as audio signals using wave would clearly lead to wrong output. But writing the raw data to file directly ensures that the header information is preserved and can be picked up by other media players. 

I'm still confused about one thing though: why the raw bytes data looked like this: b'\x9b\xf38\x83`b\xb2\x82*\x12T\xb4\x05\xc2\x0e\\\x049\xaehJ\xa6T\xa0\xc9B\xa8\t`a\n%*', shouldn't the range of byte values go from \x00 to \xff? Do you have any idea of that?

knorth55, 2020-12-17:
> I just gained more understanding of why wave wasn't able to write the correct signal while directly writing raw data to files works and I wanted to confirm it with you @knorth55. I think the reason [...]

I'm not a specialist about audio codec, but probably you are right.

>I'm still confused about one thing though: why the raw bytes data looked like this: b'\x9b\xf38\x83`b\xb2\x82*\x12T\xb4\x05\xc2\x0e\\x049\xaehJ\xa6T\xa0\xc9B\xa8\t`a\n%*', shouldn't the range of byte values go from \x00 to \xff? Do you have any idea of that?

First of all, `AudioData.data` is `uint8[]` and, in your code, you treat as byte (or string in python2) in `b''.join(frames))`.
Then, I have no idea, but maybe you wrongly encode and decode the data.
```
$ rosmsg show audio_common_msgs/AudioData
uint8[] data
```

ChanganVR, 2021-01-13:
Sorry for the late reply, as far as I know, unit8 is the same data type as byte in python. What I just figured out was values like b'\xaehJ' does not only represent a single byte value but multiple ones and here is a good explanation: https://stackoverflow.com/questions/41421501/strange-notation-for-python-3-bytes