ros-drivers/audio_common #165: AudioData messages  (audio_common_msgs) empty when echoing  /audio or subscribing to /audio in python script

ameliefroessl, 2020-12-17:
Hi, 

I'm experiencing a very weird problem. I have no idea how this is happening, but maybe you can help me out. When I echo the `/audio` topic played back either from a bag file or publishing the audiodata live,  echo will be completely empty and trying to connect to the topic via python subscriber, the callback is never even called. However, `rostopic hz /audio` indicates that there are messages coming through. Also, rqt_bag shows me there are messages in the bag. I can also rerecord what's in the bag and the recorded bag also has those messages when I check them in rqt. I can also publish and subscribe to  `std_msgs/UInt8MultiArray` messages, which have the same byte data field (uint8[]) as the AudioData messages. 

I am able to see the messages and subscribe to the topic on a different machine. I'm running Ubuntu 18 on both and installed is ROS melodic. I checked on the machine for which all of this works, audio_common_msgs 0.3.7 is installed and on the machine it does not work I have audio_common_msgs 0.3.9, if that is relevant... 

Thanks for any help !

knorth55, 2020-12-17:
can you share us the rosbag file and how to reproduce your error?

ameliefroessl, 2020-12-17:
Hi yes attached you can find the bagfile  and you can simply play the bag with `rosbag play test_1.bag` and then try `rostopic echo /audio `

[test_1.zip](https://github.com/ros-drivers/audio_common/files/5708657/test_1.zip)

knorth55, 2020-12-17:
I could echo `/audio` message with the rosbag, and I could not reproduce your problem...

```bash
$ rostopic echo -n1 /audio
data: [51, 2, 250, 1, 143, 0, 151, 0, 64, 0, 68, 255, 185, 254, 209, 254, 177, 254, 224, 255, 130, 1, 247, 0, 206, 0, 87, 0, 72, 255, 179, 255, 249, 255, 36, 0, 242, 0, 56, 1, 250, 0, 83, 1, 49, 2, 46, 2, 41, 1, 240, 255, 238, 254, 206, 254, 189, 254, 66, 254, 140, 254, 176, 255, 99, 0, 67, 0, 159, 0, 172, 0, 249, 255, 155, 0, 9, 2, 235, 2, 6, 4, 165, 4, 25, 4, 56, 4, 90, 4, 117, 3, 72, 2, 253, 0, 11, 0, 143, 255, 109, 255, 196, 255, 239, 255, 137, 255, 206, 254, 152, 254, 222, 254, 13, 255, 245, 254, 24, 254, 28, 254, 83, 255, 116, 255, 165, 254, 209, 253, 155, 253, 144, 254, 115, 254, 253, 252, 203, 252, 156, 253, 120, 253, 35, 253, 176, 252, 82, 253, 219, 254, 17, 255, 29, 255, 115, 254, 109, 254, 146, 255, 126, 255, 177, 254, 101, 254, 152, 254, 47, 254, 173, 253, 28, 254, 36, 254, 35, 254, 68, 254, 116, 253, 136, 253, 228, 253, 201, 252, 41, 252, 214, 251, 210, 250, 37, 251, 160, 252, 228, 253, 61, 255, 119, 0, 176, 0, 187, 255, 150, 254, 187, 254, 80, 0, 17, 2, 18, 4, 100, 5, 174, 5, 207, 5, 175, 4, 126, 4, 209, 5, 43, 6, 93, 6, 125, 6, 173, 5, 138, 4, 144, 3, 66, 2, 30, 2, 103, 3, 156, 3, 205, 2, 181, 1, 182, 1, 102, 2, 42, 2, 251, 1, 200, 1, 189, 0, 131, 255, 129, 254, 57, 253, 169, 252, 50, 253, 233, 252, 254, 250, 108, 249, 190, 249, 107, 251, 126, 253, 69, 254, 33, 253, 48, 253, 138, 253, 174, 252, 169, 252, 52, 253, 65, 255, 184, 1, 181, 2, 188, 3, 215, 3, 196, 2, 55, 2, 168, 1, 42, 1, 140, 1, 200, 1, 241, 0, 4, 1, 223, 1, 83, 1, 51, 0, 139, 255, 213, 255, 132, 0, 70, 1, 94, 1, 182, 255, 123, 254, 144, 254, 246, 254, 32, 255, 185, 254, 57, 254, 114, 253, 182, 253, 181, 254, 119, 254, 86, 255, 218, 255, 147, 254, 211, 255, 145, 1, 72, 1, 19, 1, 144, 1, 145, 1, 175, 1, 220, 1, 37, 1, 2, 1, 72, 0, 171, 255, 213, 255, 15, 255, 90, 254, 181, 253, 52, 253, 241, 252, 23, 253, 22, 253, 23, 252, 211, 252, 172, 253, 136, 253, 226, 254, 136, 255, 123, 0, 58, 2, 78, 3, 136, 4, 109, 5, 234, 5, 218, 5, 225, 5, 155, 5, 78, 5, 93, 5, 200, 4, 88, 4, 137, 3, 230, 1, 86, 1, 0, 1, 212, 255, 88, 255, 121, 254, 24, 253, 105, 252, 168, 251, 88, 251, 205, 251, 87, 252, 106, 252, 54, 252, 22, 252, 47, 252, 28, 253, 29, 254, 225, 253, 135, 252, 176, 251, 13, 252, 99, 253, 196, 254, 106, 255, 135, 0, 78, 1, 104, 1, 74, 1, 134, 1, 131, 2, 76, 2, 246, 1, 244, 1, 201, 1, 215, 1, 233, 1, 229, 1, 63, 2, 81, 3, 163, 2, 62, 1, 227, 255, 139, 253, 45, 253, 188, 253, 160, 252, 137, 252, 176, 253, 185, 254, 228, 254, 212, 253, 158, 253, 56, 254, 29, 254, 91, 253, 34, 253, 191, 253, 93, 254, 56, 254, 131, 254, 99, 0, 226, 1, 149, 1, 225, 0, 32, 1, 175, 1, 154, 1, 95, 0, 139, 255, 82, 0, 93, 1, 100, 1, 126, 0, 157, 255, 180, 254, 57, 254, 195, 254, 46, 255, 252, 254, 166, 254, 127, 254, 154, 255, 145, 0, 122, 0, 71, 0, 85, 255, 51, 255, 63, 0, 145, 0, 208, 0, 38, 1, 171, 1, 82, 2, 153, 2, 160, 2, 209, 2, 229, 3, 166, 4, 36, 5, 185, 5, 149, 5, 219, 5, 119, 6, 189, 6, 146, 6, 57, 6, 100, 5, 124, 4, 195, 4, 154, 5, 146, 5, 188, 4, 66, 3, 225, 0, 134, 255, 58, 255, 60, 254, 132, 252, 47, 252, 245, 252, 181, 252, 246, 252, 166, 252, 90, 252, 35, 253, 135, 253, 190, 253, 186, 253, 61, 254, 65, 254, 169, 253, 31, 254, 32, 254, 126, 253, 7, 253, 190, 252, 64, 253, 113, 253, 32, 253, 57, 252, 96, 252, 112, 254, 132, 255, 221, 255, 75, 255, 243, 252, 122, 251, 1, 251, 250, 250, 57, 252, 236, 253, 56, 254, 90, 253, 54, 253, 237, 252, 122, 252, 210, 251, 185, 250, 117, 250, 16, 252, 249, 253, 14, 254, 44, 254, 226, 253, 0, 254, 184, 255, 173, 0, 169, 1, 199, 2, 222, 2, 164, 2, 189, 2, 118, 3, 88, 3, 72, 3, 128, 4, 204, 4, 50, 5, 189, 4, 77, 3, 187, 2, 103, 2, 115, 2, 5, 2, 84, 1, 253, 255, 186, 254, 201, 254, 132, 255, 24, 0, 178, 255, 86, 255, 152, 254, 131, 253, 221, 252, 38, 252, 1, 252, 194, 251, 160, 251, 46, 252, 154, 253, 22, 0, 255, 1, 144, 3, 88, 3, 225, 1, 68, 2, 129, 2, 141, 3, 144, 5, 74, 6, 2, 7, 39, 7, 27, 7, 190, 6, 177, 5, 78, 5, 49, 5, 43, 5, 94, 5, 34, 5, 93, 4, 27, 3, 250, 0, 118, 255, 37, 254, 217, 251, 106, 250, 205, 249, 159, 248, 225, 247, 69, 248, 171, 248, 44, 249, 86, 249, 93, 249, 229, 249, 47, 250, 146, 250, 18, 251, 162, 252, 27, 254, 195, 254, 158, 0, 213, 1, 98, 1, 101, 1, 102, 2, 6, 3, 99, 3, 49, 4, 2, 5, 210, 5, 79, 6, 133, 6, 145, 6, 6, 6, 86, 5, 129, 4, 157, 3, 52, 3, 122, 2, 138, 1, 97, 1, 106, 1, 180, 0, 52, 0, 4, 0, 195, 254, 76, 253, 217, 252, 36, 253, 250, 252, 207, 251, 108, 250, 16, 250, 167, 250, 239, 249, 38, 249, 83, 250, 150, 251, 28, 253, 20, 255, 81, 0, 25, 1, 176, 1, 155, 1, 228, 1, 7, 3, 3, 3, 115, 2, 129, 1, 228, 0, 114, 1, 28, 2, 166, 2, 140, 1, 224, 0, 253, 1, 52, 2, 148, 2, 252, 2, 106, 2, 69, 2, 104, 2, 150, 1, 110, 0, 181, 255, 224, 254, 91, 255, 12, 0, 120, 255, 211, 254, 40, 254, 141, 253, 99, 253, 115, 253, 117, 253, 69, 253, 197, 252, 230, 252, 74, 253, 238, 253, 58, 254, 23, 254, 193, 254, 48, 255, 148, 255, 247, 255, 191, 255, 128, 255, 94, 255, 66, 254, 120, 253, 108, 254, 95, 254, 109, 253, 154, 253, 240, 253, 87, 255, 215, 0, 102, 1, 85, 2, 0, 2, 25, 1, 212, 1, 123, 2, 122, 1, 28, 1, 141, 0, 139, 255, 148, 255, 214, 253, 152, 252, 205, 252, 6, 252, 128, 252, 210, 253, 75, 254, 83, 254, 227, 253, 179, 254, 167, 255, 100, 0, 248, 0, 236, 0, 230, 0, 144, 0, 18, 0, 49, 255, 157, 254, 235, 254, 207, 255, 159, 0, 177, 0, 187, 0, 177, 1, 65, 3, 46, 3, 79, 1, 67, 1, 191, 1, 102, 2, 55, 4, 142, 3, 89, 2, 29, 4, 255, 4, 63, 4, 250, 3, 190, 2, 83, 2, 16, 2, 17 [... cut ...] , 230, 1, 105, 1, 126, 1, 81, 2, 38, 3, 93, 2, 162, 1, 234, 1, 158, 2, 216, 1, 156, 255, 224, 253, 111, 253, 82, 253, 11, 252, 182, 251, 76, 251, 198, 248, 190, 246, 66, 246, 213, 245, 61, 246, 180, 246, 140, 247, 85, 249, 246, 249, 22, 251, 3, 252, 97, 251, 33, 252, 204, 254, 53, 1, 112, 3, 233, 3, 244, 2, 74, 4, 217, 5, 36, 6, 105, 7, 195, 7, 34, 8, 186, 8, 95, 7, 0, 7, 88, 7, 59, 6, 215, 4, 71, 3, 117, 1, 193, 1, 175, 2, 253, 1, 171, 0, 244, 254, 235, 254, 247, 255, 141, 255, 1, 254]
---

```

ameliefroessl, 2020-12-17:
Ok thanks for trying... I think I installed audio_common via : sudo apt-get install ... Do you have any ideas what might cause this issue?.. I'm slowly running out of ideas...

knorth55, 2020-12-18:
sorry, but i have no idea...

ameliefroessl, 2020-12-19:
Ok So i think i found where the problem comes from. I tried it again on a different machine with the same setup. Ubuntu 18 and ros melodic. I installed audio_common through sudo apt install ros-melodic-audio-common. Which gave me the same problem. Then I uninstalled it and built it from source and then it worked. I am not sure what the exact reason was but just so you know...

knorth55, 2020-12-19:
I installed `audio_common` `0.3.9` with `apt`, but I cannot reproduce the error.

ameliefroessl, 2020-12-21:
ok you said you are using kinetic.. i'm using melodic. Maybe that's what's causing the problem?

knorth55, 2020-12-21:
maybe it is not the reason.