ros-drivers/audio_common #166: How to mute soundplay_node?

5tan, 2021-01-21:
This is bug or feature request.

I would like to add "mute" capability to robot using [ReSpeaker](https://wiki.seeedstudio.com/ReSpeaker_Mic_Array_v2.0/) as audio output.

Unfortunately system mute button doesn't work when I use soundplay and reaspeaker. This is weird because when I use sound_play with sound card on my laptop, mute button works. Also if I use ReSpeaker with other audio apps (e.g. chromium/YouTube) mute button works.

Summary:
default laptop sound card + sound_play -> mute button works 
ReSpeaker + sound_play -> mute button doesn't work 
ReSpeaker + chromium/YouTube -> mute button works 

Any idea what is the issue here? Is this bug in ReSpeaker firmware? In sound_play? Or in some gstreamer plugin? How to verify this?

What option do you recommend to mute soundplay_node? Do you find "mute" a useful feature to add?

knorth55, 2021-01-21:
can you check if you can change the volume level of respeaker by`alsamixer`?
if not, we cannot change it, probably.
Can you also try `volume=0` field in `SoundRequest`?

5tan, 2021-01-22:
with alsamixer no, "This sound device does not have any controls."
[image]
but with `pavucontrol` yes, I can change volume and mute. It works when I use e.g. chromium, but doesn't work when I use sound_play. What's the difference?
[image]

> Can you also try volume=0 field in SoundRequest?

Yes, this can be workaround, but not perfect in distributed system (since sound requests might come from many source nodes).

knorth55, 2021-01-22:
`soundplay_node` uses `alsasink`, which relies on `alsa`.
so if you cannot control volume your device from `alsa`, you cannot control it, probably (I'm not a `alsa` specialist, so I'm not 100% sure).
https://github.com/ros-drivers/audio_common/blob/master/sound_play/scripts/soundplay_node.py#L85

I found out `respeaker` alsa issue in github, so please ask about `alsa` support in the issue.
https://github.com/respeaker/usb_4_mic_array/issues/39

Or another solution is to different `sink` in `soundplay_node.py`.

5tan, 2021-01-22:
Thanks @knorth55 !
Can I change sink by configuration? Or do I have to modify soundplay_node code? https://github.com/ros-drivers/audio_common/blob/master/sound_play/scripts/soundplay_node.py#L85

knorth55, 2021-01-22:
you need to change the source code following part.
I'm not sure it will work if you simply change just one line.
https://github.com/ros-drivers/audio_common/blob/master/sound_play/scripts/soundplay_node.py#L85

5tan, 2021-01-25:
As it turned out, I just had to install `gstreamer1.0-pulseaudio` (and configure my docker for pulseaudio access). `soundplay_node` began to use pulseaudio instantly (no changes in code required). 

I am not sure how this works (how alsasink got promoted into pulsesink?), but it's fantastic finding for ReSpeaker users.

For me the case is closed, thanks!

closed 2021-01-25 by 5tan

yuetin, 2024-05-29:
> As it turned out, I just had to install `gstreamer1.0-pulseaudio` (and configure my docker for pulseaudio access). `soundplay_node` began to use pulseaudio instantly (no changes in code required). [...]

Could you explain how to change the docker archive?
i have same problem in this