ros-drivers/audio_common #177: ROS2 migration?

airballking, 2022-02-02:
Dear maintainers of audio_common,

quick question: Are there any plans for migrating this package to ROS2? 

If yes, what is the agenda? Do you need help from the community?

If no, what is the main reason? Potential reasons that I can come up with: Technical difficulty, exising (ROS2) alternative solution, or lack of resources.

Please consider also answering here: https://answers.ros.org/question/354035/audio_common-for-ros2/

Thanks a lot for your efforts and support!
Georg.

knorth55, 2022-02-02:
@airballking 
This is not related to `audio_common`, but are you using ROS2 in IAI bremen?
Because you have PR2 (indigo?), how do you use ROS2?
We have PR2 in our lab, and I want to know how you use ROS2 and PR2.

knorth55, 2022-02-14:
@airballking I didn't know that you and Ferenc made a start-up.
It is cool, i want to see your products in near future!
I will ask PR2 upgrade to IAI by myself, thank you.
Also, I can support or participate ROS2 porting, too!

knorth55, 2022-02-15:
I made a PR for this issue. #176 
It is still WIP (mainly `sound_play`).

knorth55, 2022-02-22:
I made ROS2 branch, so please check it!
https://github.com/ros-drivers/audio_common/tree/ros2

Patrick-AA, 2022-05-24:
Hello,
I am currently testing the ROS2 branch as I might need it in a project soon.
The packages work well, however the "ns" no longer exists for "group" in the launch files. I switched it to "push-ros-namespace".

knorth55, 2022-05-24:
@Patrick-AA thank you for your report. can you make a PR to `ros2` branch?

BrettRD, 2022-11-11:
Hi @knorth55, I only saw your February post on ros-answers today

I'm planning a development push to align https://github.com/BrettRD/ros-gst-bridge and https://github.com/clydemcqueen/gscam2 allowing video IO on composable nodes

gst_bridge supports audio very well now, and I'm interested in bringing some of that functionality to audio_common but I'm not sure where to start

knorth55, 2022-11-11:
@BrettRD 
I made a development branch for ROS2, so please check this.
https://github.com/ros-drivers/audio_common/tree/ros2

If you want to add new functions, please make a PR to `ros2` branch.