ros-drivers/audio_common #180: Can't play sound when self-starting roslaunch without login in ubuntu16.04

zhaozhi1995, 2022-02-28:
I startup the "soundplay_node.launch" with systemctl service,the soundplay node will not play sound successfully. But when I login to ubuntu16.04's desktop, the node will work. 

Why it's not work without login to desktop,and how to fix the problem?

knorth55, 2023-02-07:
Now I found out this issue comes from `pulseaudio`.
`pulseaudio` service starts when user logs in, so that `sound_play` cannot make sound.
In order to run `sound_play` without login, we need to run `pulseaudio` in system wide mode, and set `XDG_RUNTIME_DIR=/var/run` in your systemd service.
in system wide mode, we need to change config in `/etc/pulse/` properly and now I solve the issue.

knorth55, 2023-02-07:
please check #219 for more detailed info

closed 2023-02-07 by knorth55