ros-drivers/audio_common #189: scripts/play.py does not work for absolute path

k-okada, 2022-05-25:
```
audio_common/sound_play/scripts$ ./play.py $(pwd)/../sounds/say-beep.wav 1.0
[INFO] [1653473671.097828]: Playing "/home/k-okada/catkin_ws/ws_audio/src/audio_common/sound_play/scripts/../sounds/say-beep.wav".
```
raise error like
```
process[sound_play/is_speaking-3]: started with pid [982]
[INFO] [1653473766.379425]: Loading from plugin definitions
[INFO] [1653473766.886719]: sound_play node is ready to play sound
[ERROR] [1653473767.063060]: Exception in callback: 'SoundPlayNode' object has no attribute 'filesound'
[INFO] [1653473767.068195]: Traceback (most recent call last):
  File "/home/k-okada/catkin_ws/ws_audio/src/audio_common/sound_play/scripts/soundplay_node.py", line 192, in callback
    sound = self.select_sound(data)
  File "/home/k-okada/catkin_ws/ws_audio/src/audio_common/sound_play/scripts/soundplay_node.py", line 109, in select_sound
    sound = self.filesound[data.arg]
AttributeError: 'SoundPlayNode' object has no attribute 'filesound'
```
which was working on 0.3.13.

closed 2022-05-25 by knorth55

knorth55, 2022-05-25:
sorry, I fixed the issue in #190

knorth55, 2022-05-25:
Yes, I will. thats my pending task.

PR #190 (2022-05-25): fix typo in soundplay_node.py
fix #189