ros-drivers/audio_common #194: Frame Duration

encrxpted, 2022-06-30:
Hi, is there any way to control the length of each frame that is sent to the audio rostopic? I am trying to integrate ROS with WebRTCVAD, which only accepts frames of 10, 20, or 30ms. Thanks!

knorth55, 2022-07-01:
can you try `throttle` node to stabilize the topic frame rate?
http://wiki.ros.org/topic_tools/throttle

encrxpted, 2022-07-01:
> can you try `throttle` node to stabilize the topic frame rate? http://wiki.ros.org/topic_tools/throttle

I think that throttle will still send over the same messages, right? Won't each frame will still be the wrong duration since throttle just republishes the ros messages?

knorth55, 2022-07-02:
yes, but if /audio topic ha s higher frequency, it is worth trying.
but the sound may not be continuous.
 if you are talking about audio_capture node, the frequency of the node loop is controlled by gstreamer, and we currently don't support the frequency change.

encrxpted, 2022-07-05:
> 

Then is there any way for me to "resize" the chunks after I receive the audio data?

knorth55, 2022-07-05:
currently no. we need to implement it.

encrxpted, 2022-07-05:
Adding to the above:
We are able to bag the audio data and save it as a wav file. Plus, I am able to take audio data from a wav file, splice it into the chunk sizes I need, and successfully use the VAD library to process those chunks. Thus, I know that you can certainly "resize" the chunks via the wav file. My question now is, how can I directly just change the chunk size of the audio data when I receive enough bytes?

knorth55, 2022-07-06:
I have no idea, sorry.
I think just concatenating several topics is enough, but I'm not sure.