ros-drivers/audio_common #196: High latency of audio_capture audio stream

reisenhofer, 2022-08-12:
I am working on a project where we use live transmissions of an audio stream captured from a microphone (via audio_capture) and a video stream from a webcam (via video_stream_opencv). We work with Python and ROS Noetic on Ubuntu 20.04 and use audio_play and image_view, respectively, to playback the streams at a remote location.

Compared to the video stream, the audio_capture/audio_play audio stream has a latency of more than 1 (maybe even 2) seconds. To me, this seems unnecessarily high, in particular as the audio stream should require much less bandwidth than the video stream. Can you think of any possibility of reducing this latency when working with audio_capture/audio_play?

Thanks so much in advance!

knorth55, 2022-08-18:
I have no idea.
can you check if the high latency also occurs in alsa or non-ROS programs?

pourya-shahverdi, 2022-09-07:
Check your local network bandwidth. I moved to Ethernet, and the latency is much better.