ros-drivers/audio_common #197: Question about usage

HaoLiRobo, 2022-08-23:
I read the audio_capture documentation, but didn't find how to deal with the args in launch files. For example, if I want to record a wav file with channel 2, what will the data be like in the topic /audio? Will it be a 2 dim array? 

Thank you so much for making this wonderful package!

knorth55, 2022-08-23:
audio data is formatted in wav or mp3 style.
that format can carry multi channel data, too.
however audio_play and audio_capture should share the # of channel information beforehand.
 or we can send the meta information with `AudioInfo` msg.

HaoLiRobo, 2022-08-23:
Thanks a lot! I managed to capture audio with wave format, channel 2 and sample rate 44100. Then I ran audio_play, which sounds fine. But when I try to save the data from /audio topic as a wav file, it sounds pretty awful. I think I miss something important. What I did was,

self.complete_audio = []

def audio_callback(self, data):
    audio = np.frombuffer(data.data, dtype=np.uint8).tolist()
    self.complete_audio.extend(audio)

When complete recording, I save it as:
    deinterleaved_audio = np.array([audio_buffer[::2], audio_buffer[1::2]], dtype=np.uint8).T
    write('test.wav', SR, deinterleaved_audio)

I also plot the audio data for each channel:

[image]
[image]

Sorry that I don't know much about this domain, so I might miss something to correct address the data. Thank you so much again for helping me out!

HaoLiRobo, 2022-08-24:
Thanks a lot! Do I have to save to file to get the listenable audio? Can I directly can a two dim vector with each store the sound for each channel? I have some real-time demands, so I'm just afraid, if I save it to file and than read it, it will be too slow.

HaoLiRobo, 2022-08-24:
Nvm, I tried to figure out that the first 24 numbers of the wav data is the meta information, so I just simply remove the first 24, the msg.data would be the listenable audio data.

If anyone want to get the wav data, just do the following:
audio = np.frombuffer(msg.data, dtype=np.int16)[24:].tolist()

Based on your sample rate and channels, you can decode it as normal wav file.

Cheers!

knorth55, 2022-08-24:
Good to hear that!

closed 2022-08-24 by HaoLiRobo