ros-drivers/audio_common #198: sound_play_plugin.yaml is not included in .deb

tokoro10g, 2022-08-28:
```
[INFO] [1661653504.631605] [/sound_play:rosout]: Loading from plugin definitions
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/sound_play/soundplay_node.py", line 456, in <module>
    SoundPlayNode()
  File "/opt/ros/noetic/lib/sound_play/soundplay_node.py", line 361, in __init__
    with open(plugin_yaml) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/opt/ros/noetic/share/sound_play/sound_play_plugin.yaml'
```

I looked into the deb package, but
```
$ dpkg -c ros-noetic-sound-play_0.3.14-1focal.20220818.173544_amd64.deb | grep yaml
```
returned nothing.

I believe this is a packaging issue. Could you fix this?

knorth55, 2022-08-28:
Thank you for reporting.
Sorry for the mistake.
I fix the problem in #199.
I will release the fixed version soon.

knorth55, 2022-08-29:
waiting for the release https://github.com/ros/rosdistro/pull/34239 https://github.com/ros/rosdistro/pull/34240

knorth55, 2022-09-06:
https://github.com/ros/rosdistro/pull/34239 and https://github.com/ros/rosdistro/pull/34240 are merged.

@tokoro10g 
are you using `ros-tesing`?
if so, can you upgrade `sound_play` and  check if the issue is solved?

tokoro10g, 2022-09-07:
@knorth55 
Confirmed that this is fixed in `0.3.15-1focal.20220829.221132`.
Thank you so much for your quick response!

closed 2022-09-07 by tokoro10g

knorth55, 2022-09-07:
Thank you for checking!