ros-drivers/audio_common #201: No time in the messages

v4hn, 2022-09-07:
Hey @knorth55 , I didn't know you maintain this nowadays :+1:  :smile_cat: 

I work on an audio application right now where dropped samples and accurate times are important.
`AudioData` does not have a header/stamp/serial number, so there is no way to detect lost samples on the subscriber side.
The best solutions I found online subscribe to `AudioData` and put it in a custom message that includes a stamp filled in with `Time::now()` which is of course nonsense.

I wonder whether others have problems with this. I mean, changing the message for everyone would sadly be a rather drastic (but from my perspective justified) change...

knorth55, 2022-09-08:
I agree with the idea.
`MessageFilter` needs `stamp` data, so I'm also thinking of adding `timestamp` info in the message.
My opinion is adding `AudioDataStamped` message, to keep backward compatibility.
IMO, header data is huge for micro controller, so `AudioData` is useful in that case.

knorth55, 2022-09-08:
Changes:
- [ ] add `AudioDataStamped`
- [ ] publish both `AudioData` and `AudioDataStamped` in `audio_capture`
- [ ] subscribe both `AudioData` and `AudioDataStamped` in `audio_play`

v4hn, 2022-12-22:
You implemented the important parts in #202  and #204 . Thanks :)

closed 2022-12-22 by v4hn