ros-drivers/audio_common #209: Convert rclcpp nodes into components

weeshal, 2022-10-25:
Feature Request:

Convert the audio_capture and audio_play nodes into [rclcpp components](https://docs.ros2.org/latest/api/rclcpp_components/). The benefits this would have, while still maintaining backwards compatibility (non-functional change) are:

1. Allow for [intra-process communication](https://docs.ros.org/en/foxy/Tutorials/Demos/Intra-Process-Communication.html) when nodes are running in the same component container - removing the copy overhead between publishers and subscribers in the same container (think capture and play running the same container with less latency - especially important for relatively heavy audio msg)
2. Allow for the nodes to be [loaded/unloaded dynamically.](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Composition.html) For my current use-case, we want to record audio files for n seconds, instead of using a separate node to capture /audio and save to a wav file, we can utilize the already implemented gstreamer pipeline in this package to run the node for n seconds and save a wav file in this manner.

@knorth55 let me know if this is something you would be interested in - I can make the change, test and submit a PR if so!

knorth55, 2022-10-26:
@weeshal If you can, I look forward to seeing your new feature!

knorth55, 2023-01-19:
Done in #215

closed 2023-01-19 by knorth55