ros-drivers/audio_common #232: ros2 branch is not compiling for ROS 2 Humble

Noel215, 2023-05-31:
This is the error when compiling in ROS 2 Humble:

```
Starting >>> sound_play
--- stderr: sound_play                                                                      
CMake Error at /opt/ros/humble/share/ament_cmake_python/cmake/ament_python_install_package.cmake:106 (add_custom_target):
  add_custom_target cannot create target
  "ament_cmake_python_symlink_sound_play" because another target with the
  same name already exists.  The existing target is a custom target created
  in source directory
  "/home/user/exchange/alum/my_ws/src/audio_common/sound_play".  See
  documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_python/cmake/ament_python_install_package.cmake:39 (_ament_cmake_python_install_package)
  CMakeLists.txt:36 (ament_python_install_package)

CMake Error at /opt/ros/humble/share/ament_cmake_python/cmake/ament_python_install_package.cmake:141 (add_custom_target):
  add_custom_target cannot create target
  "ament_cmake_python_build_sound_play_egg" because another target with the
  same name already exists.  The existing target is a custom target created
  in source directory
  "/home/user/exchange/alum/my_ws/src/audio_common/sound_play".  See
  documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_python/cmake/ament_python_install_package.cmake:39 (_ament_cmake_python_install_package)
  CMakeLists.txt:36 (ament_python_install_package)

---
Failed   <<< sound_play [1.17s, exited with code 1]

```

knorth55, 2023-05-31:
plase check this pr

https://github.com/ros2/rosidl_python/pull/187

Noel215, 2023-06-01:
Good to know! 

Then, are you releasing the packages for humble and following distributions after that is merged?