ros-drivers/audio_common #235: ros2 branch has ros1 headers

ashBabu, 2023-08-08:
Hi, I would like to use `sound_play` in `ros2` C++. The aim is that when an obstacle is detected (ros2 subscriber), provide an audio alert to the user. 
I followed the [this](https://github.com/ros-drivers/audio_common/issues/227#issuecomment-1479679415) and got it to work by 
`ros2 run sound_play soundplay_node.py` and `ros2 run sound_play say.py "Hello"`. However when I do,

```
#include <sound_play.h>

class sound
{
sound_play::SoundClient sc_;
}
```
this starts complaining about `ros1` headers.
Any fixes for this?

knorth55, 2023-08-08:
I haven't implemented `sound_play.h` into `ROS2` yet.
Your contribution is welcome, but I recommend to use `python` version.

BTW, can you paste full error log?

ashBabu, 2023-08-08:
Hi, thanks for the reply. I have removed all the sound_play stuff from my program and hence dont have a full error log now. From memory, I can say that these [lines](https://github.com/ros-drivers/audio_common/blob/ros2/sound_play/include/sound_play/sound_play.h#L41-L43) were the main causes and wherever `ros::Nodehandle` is used

knorth55, 2023-08-08:
I think including `sound_play.h` for `ros2` is useless, because it doesn't support ros2....

ashBabu, 2023-08-08:
I am actually migrating to ros2 from ros1. The sound_play.playWave() really worked for me in ros1. So was just trying to replicate the same in ros2. 
Any other suggestions are welcome

antbono, 2023-09-05:
Thank you for this issue... I'm trying to compile sound_play in ros2 rolling but i got this error

```
CMake Error at /opt/ros/rolling/share/ament_cmake_python/cmake/ament_python_install_package.cmake:141 (add_custom_target):
  add_custom_target cannot create target
  "ament_cmake_python_build_sound_play_egg" because another target with the
  same name already exists.  The existing target is a custom target created
  in source directory
  "/home/toto/Gdrive/uni/robocup/robocup_ws/src/audio_common/sound_play".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /opt/ros/rolling/share/ament_cmake_python/cmake/ament_python_install_package.cmake:39 (_ament_cmake_python_install_package)
  CMakeLists.txt:36 (ament_python_install_package)
```
I think because `sound_play.h` is only for ros1

knorth55, 2023-09-05:
@antbono no.
your issue is different and it is reported in #227.
please check #227 to avoid the error.