ros-drivers/audio_common #239: Question/Tutorial : how to effectively use audio_capture_node in ROS2

antbono, 2023-09-05:
Hello everyone!
I am quite new to ros2 and I am trying to understand how to effectively use `audio_capture_node`.
I think that a common case is recording when some conditions are true and, possibly, for a variable period of time.
How can you do this? Do you launch this node from another one? If yes, how?
I hope the question is clear.

Thanks!

knorth55, 2023-09-05:
please read launch files and figure out how to use it.
you may be need to use `audio_play`, too.
https://github.com/ros-drivers/audio_common/tree/ros2/audio_capture/launch

weeshal, 2023-10-05:
@antbono You can look into [ros2 components.](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Composition.html) The capture node was recently written as a component so you can load and unload it as necessary. The link I have attached uses the command line interface, but you'll need to use ros2 services to emulate a similar workflow in rclpy/rclcpp.

closed 2026-05-15 by knorth55