ros-drivers/audio_common #241: ROS2 Branch: SoundClient blocking mode does not work

antbono, 2023-09-17:
Hello Everyone!

I'm using the ros2 branch. I tried to execute the simple node in `scripts/play.py`  and I noticed that the blocking mode is not working actually: the program ends before the sound file is completely played.

Has anyone had the same issue?

Thanks

knorth55, 2023-09-19:
could you check if the client topic stops at this line?
https://github.com/ros-drivers/audio_common/blob/e4b70876edf74052ca776f00158317a277b0d5af/sound_play/sound_play/libsoundplay.py#L377

antbono, 2023-09-21:
Hi @knorth55!

I checked out, and I can confirm that the client stops at the line you highlighted. 

Moreover when I use the non blocking mode I get this kind of error

`[ERROR] [1695310239.114727197] [sound_play]: callback: sound_play.msg.SoundRequest(sound=-2, command=1, volume=1.0, arg='/tmp/output.ogg', arg2='')
`

but I can actually listen to the audio file.

knorth55, 2023-09-24:
so you mean that this issues is solved?

antbono, 2023-09-25:
No, the issue is not solved.
Even if sound_play_node.py send a message of action completed

https://github.com/ros-drivers/audio_common/blob/e4b70876edf74052ca776f00158317a277b0d5af/sound_play/scripts/soundplay_node.py#L560

 when I use  SoundClient in my node, it never finish the command 

https://github.com/ros-drivers/audio_common/blob/e4b70876edf74052ca776f00158317a277b0d5af/sound_play/sound_play/libsoundplay.py#L377

and thus my node waits the response forever.