ros-drivers/audio_common #242: Add audio common ros2 branch to `iron` distribution

jackiescanlon, 2023-11-01:
Hello, I would like to be able to install audio common packages via `rosdep` on Iron. However it looks like audio common ros2 barnch hasn't been added to https://github.com/ros/rosdistro/blob/master/iron/distribution.yaml yet. I was wondering if it could be added?

knorth55, 2023-11-01:
Hi,
I want to release it, but there is a big issue unsolved in ROS2.
We are waiting for the issue solution, and I posted the solution but declined...
So we are currently unable to release the repo. Sorry for inconvenience.

https://github.com/ros2/rosidl_python/issues/141

jackiescanlon, 2023-11-01:
Ah yes, I have actually looked at the issue you linked and your PR before because we faced the same issue in our code base! I forgot you had mentioned it was affecting `sound_play`.

Thanks for your response and for that PR, I really hope they fix that because it is a big pain.

rkent, 2025-08-24:
I've been assisting with a PR https://github.com/ament/ament_cmake/pull/587 that should alleviate this issue. It's been going on for months, but hopefully will be resolved soon.

knorth55, 2025-08-28:
@rkent That's great! Thank you so much!

knorth55, 2025-12-26:
@rkent 
I start working on this issue, too.
My plan is to split and add new package `sound_play_msgs`.
https://github.com/ros-drivers/audio_common/pull/264

I can test build on CI, but I haven't prepared ROS2 environment.
Could you help me trying the PR on ROS2 environment when you have time?

rkent, 2025-12-31:
I'm planning on helping you with this, but I have been busy over the holidays. I'll be much more available after Jan 5.

knorth55, 2026-02-26:
@severin-lemaignan 
Hi, actually there is one issue related `ament_cmake` update.
I'm working on it, and I can release the package after that.